iNavFlight/inav #1139: Servo output broken in 857b3603c41e5faa8eb7f3c7a2b87f50f7143494

DzikuVx, 2017-01-19:
Servo output is completely broken in 857b3603c41e5faa8eb7f3c7a2b87f50f7143494

* MSP_SERVO frame reports 0 on all servo channels
* oscilloscope shows flat line on all servo lines on AirbotF4 and Omnibus targets

oleost, 2017-01-19:
Can confirm issue on spracingf3 target using servo gimbal feature.

Looking at the "Motor" tab everything is at zero( 1500 is normal ) and looking at "Servos" tab CH1 is selected on all servos, and cannot change it. (Normal is none selected and then select the aux of choice ) Tried both Master configurator and 1.5.2, but the issue is with firmware 1.6

[dump 1,5.txt](https://github.com/iNavFlight/inav/files/717635/dump.1.5.txt)
[Dump 1,6.txt](https://github.com/iNavFlight/inav/files/717636/Dump.1.6.txt)
[image]
[image]
[image]
[image]

digitalentity, 2017-01-20:
Can confirm on latest `development` + airplane mixer. Servos are stuck at 1500 - no reaction to PID output or sticks.

digitalentity, 2017-01-20:
Bisecting narrowed this down to bf46c4809f924392322b3997a2de4a04b1b1e4d5. @martinbudden can you have a look?

digitalentity, 2017-01-20:
Hm... I can no longer reproduce the issue on latest `development`. Looks like I was barking up the wrong tree here.

oleost, 2017-01-20:
@martinbudden Tried latest development today. Still no go on servo gimbal function. Motor tab is showing 1500us now however, but in the "Servos" Tab applying control over Servo 0 on Aux4 for example does not stick.

[dump and bootlog.txt](https://github.com/iNavFlight/inav/files/718848/dump.and.bootlog.txt)

stronnag, 2017-01-20:
Perhaps worth noting that servos have been broken in betaflight 3.1 for ages in case this is a result of a bf catch up.

oleost, 2017-01-28:
@martinbudden Did retest today. 
Still no go at servo gimbal feature.

martinbudden, 2017-01-29:
OK, I'll try and look at this later in the week.

digitalentity, 2017-02-05:
@martinbudden did you by any chance look at this issue?

martinbudden, 2017-02-05:
@digitalentity , no I haven't had a chance to look at this. Can you do this?

digitalentity, 2017-02-05:
I'm afraid that don't have SPRacingF3 target to test. I'll have a dig into my FC box.

digitalentity, 2017-02-07:
Tested on CC3D - works fine (both QuadX and Flying wing config).
@oleost can you re-test and if issue persists for you - open a new one for your specific configuration?

closed 2017-02-07 by digitalentity

oleost, 2017-02-07:
@digitalentity it's was specifically servo gimbal feature that did not work on spracingf3.

I have screwed together my quad again without servo to test this more. If someone else have an loose spracingf3 and a servo it would spare me some work testing this again.

DzikuVx, 2017-02-07:
Also tested for SERVO_GIMBAL for SPRF3. Works in 5c2f2f9af333685d24da7a4187fab8ada4019b9e

oleost, 2017-02-07:
Great, probably been fixed someplace