iNavFlight/inav #1143: Mixer and CLI diff bugs

digitalentity, 2017-01-20:
Not sure if these are different bugs or one bug.

Steps to reproduce - latest `development`, `CC3D_PPM` target.

 1. Install with full erase
 2. Go to CLI, type `diff`, get this output:

```
# diff
# INAV/CC3D_PPM1 1.6.0 Jan 20 2017 / 07:45:14 (cab3708)

feature RX_PPM
set acc_hardware = MPU6000
set mag_hardware = NONE
set baro_hardware = NONE

profile 0

rateprofile 0
```

 3. Go to GUI (this is important), change mixer type to AIRPLANE
 4. Go to CLI, type `diff`:

```
# diff
# INAV/CC3D_PPM1 1.6.0 Jan 20 2017 / 07:45:14 (cab3708)
mixer QUADX

feature RX_PPM
set acc_hardware = MPU6000
set mag_hardware = NONE
set baro_hardware = NONE
set 3d_deadband_throttle = 1000

profile 0

rateprofile 0
```

Note that in the `diff` output the mixer stays `QUADX`. Also note that `3d_deadband_throttle` somehow changed to 1000 (default 50).

digitalentity, 2017-01-20:
The `3d_deadband_throttle` is caused by mixed up parameter count in MSP_3D, MSP_SET_3D, MSP_RC_DEADBAND and MSP_SET_RC_DEADBAND messages and that bug is there for god known how long.

We need to decide if we want to make `3d_deadband_throttle` part of `MSP_3D` message (which is logical), or align with Betaflight where `3d_deadband_throttle` is part of `MSP_RC_DEADBAND` message.

martinbudden, 2017-01-20:
@digitalentity , can you try with https://github.com/iNavFlight/inav/pull/1144

`3d_deadband_throttle` not fixed, but mixer is. Also some other improvements. I've not merged it in yet because there is something strange going on with `systemConfig` that I haven't tracked down yet.

digitalentity, 2017-01-31:
Ok, all fixed now.

closed 2017-01-31 by digitalentity