iNavFlight/inav #1150: rssi_ppm_invert not working

sjobbe, 2017-01-22:
It seems CLI command **rssi_ppm_invert** does not do anything. Regardless if it is ON or OFF i get the inverted result: no signal 100% and vice versa, radio next to the copter powered on i get 6% signal.

To replicate the issue: 
- SP Racing EVO with PWM RSSI pads soldered together.
- Cable from X4R-SB RSSI pad to SP3 EVO RSSI input
- iNAV version 1.5.1

SPlissken, 2017-02-02:
I can confirm this issue with a D4R-II and iNAV 1.5.1:

Direct connection from receiver RSSI port to SP Racing Evo RSSI PWM input.
RSSI reading in iNAV Configfurator 1.5.2 shows as inverted: 3% for full signal, 100% for no signal.

digitalentity, 2017-02-05:
That's actually not a bug. `rssi_pwm_invert` is applied only for RSSI injected into a RC channel. We'll replace it with rssi_invert `setting` that will be applied always

digitalentity, 2017-02-05:
@sjobbe @SPlissken can you test #1234

SPlissken, 2017-02-05:
I can confirm that 'set rssi_invert = ON' now inverts the RSSI reading with a PWM input on the SP3 Racing EVO board with the current development branch.

It doesn't seem massively accurate though so some form of offset in addition to the RSSI scaling may be needed to get it matching the FrSky readings. Should I open a feature request?

Good work on getting the inversion sorted for all RSSI inputs, thanks!

digitalentity, 2017-02-05:
>It doesn't seem massively accurate though so some form of offset in addition to the RSSI scaling may be needed to get it matching the FrSky readings. Should I open a feature request?

Yes please, we'll try to have this sorted out in 1.7

closed 2017-02-05 by digitalentity

konraddi, 2018-02-14:
set rssi_ppm_invert is no valid command in iNav? I cannot invert the ezuhf ppm rssi :(

digitalentity, 2018-02-15:
@konraddi RSSI inversion is working fine in current development. Fix will be included in next release