iNavFlight/inav #1161: Parameter Groups - diff does not include mmix

stronnag, 2017-01-24:
`diff` does not include `mmix` when needed (regression against pre-PG) 2725e6a9fc09eceb81b4be6cd2a6ad3929334cbe
````
# mixer
mixer CUSTOM

# servo
```` 
pre-PG (e.g.):
````
# mixer
mixer CUSTOM
mmix reset
mmix 0  1.000 -0.625  1.000 -1.000
mmix 1  1.000 -0.625 -0.250  1.000
mmix 2  1.000  0.625  1.000  1.000
mmix 3  1.000  0.625 -0.250 -1.000

# servo

````

martinbudden, 2017-01-24:
OK, thanks. I'll look into it.

stronnag, 2017-01-24:
Thanks. I've now configured two machines via the PG CLI  (deadcat quad and a tri-copter) and tether-tested them. Seem to respond as expected. I'll fly 1.6 at the weekend.

martinbudden, 2017-01-26:
Looks like there may be a problem in the diff code for mmix that was ported over from betaflight, see https://github.com/betaflight/betaflight/issues/2223

martinbudden, 2017-01-26:
Fixed in PR https://github.com/iNavFlight/inav/pulls

closed 2017-01-26 by martinbudden