iNavFlight/inav #1172: 1.6.0 Test Log

stronnag, 2017-01-27:
It seems to me it would be useful to have a single place to record incremental test successes (and failures) on the road to next release. It may also help catch regressions. If you disagree, please close. This does not preclude obvious bugs being logged under separate issues for triage and rectification.

stronnag, 2017-01-27:
**Build Tested** : 2df6fae67458d4f2ce8c5bab7b1aff4fe985c169
**Hardware**: 
* SPRF3 on 260mm dead cat, M8N, BMP280.
* SPRF3 (Dodo) on 280mm tricopter, not equipped for navigation.

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH

**Special Points**:
* Configured via CLI `diff all`
* Default (racer) notch settings from latest configurator.

**Results**:
* Perfect 
* No regressions for bad behaviours noted.

martinbudden, 2017-01-27:
Great results. I agree on a single place for test logs. This issue can serve that purpose.

digitalentity, 2017-01-28:
Great results! @stronnag thanks for testing. I agree - one thread for test flights and logs is great (and new issues per specific bugs found)

stronnag, 2017-01-28:
**Build Tested** : 6883f08e794d8a04fc8a27cad54266770ae50807 
**Hardware**: 600mm tricopter , SPEVO, M8N
**Modes**: 
* Free flight (horizon, LOS)
* WP, PH, RTH

**Special Points**:
* Configured via CLI `diff all`
* Default (450-600) notch settings from latest configurator.

**Results**:
* Very smooth. In particular maintaining altitude on navigation transitions (WP legs, RTH).
* No regressions or bad behaviours noted.

oleost, 2017-01-29:
**Build Tested** : 3b7620c80847c03e95ca197ca58292fda7101b6c
**Hardware**: Z84 Fixed-wing, Spracing f3 mini. BN880 M8N, Frsky X4R and smartport telemetry.
**Modes**: 
* Bench test. Passtrough, Horizon and rate
* Tested Channel forwarding to Servo midpoint setting for trim.

**Special Points**:
* Configured from Preset 600mm fixedwing.

**Results**:
* Not test flown yet, but everything seems to work as expected. Looking forward to trying it out in field.

ukmook, 2017-01-29:
Build Tested : 3b7620c
Hardware: Sparky2, QAV 250, NAZA GPS, SBUS, 

Not flown yet. MAG fails, it did this on 1.5.1 / 1.5 as well

CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611, MAG=AK8963
Sensor status: GYRO=OK, ACC=OK, MAG=FAILING, BARO=OK, SONAR=NONE, GPS=NONE
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 1798, max available config: 131072
System load: 1, cycle time: 2002, PID rate: 499, RX rate: 49, System rate: 39

digitalentity, 2017-01-30:
I have a `SPARKY2` board on the way to me. Soon we'll have this MAG issue fixed (hopefully in 1.6)

stronnag, 2017-01-31:
**Build Tested** (build / flash only) : INAV/SPRACINGF3 1.6.0 Jan 31 2017 / 19:42:23 (2c3a1d04)

Unexpected line appears in diff and dump:
````
VALUE rangefinder_hardware ERROR
````
This happens even after an explicit:
````
# set rangefinder_hardware = NONE
rangefinder_hardware set to NONE
# save
````

digitalentity, 2017-02-01:
@stronnag fixed. See https://github.com/iNavFlight/inav/pull/1210

Redshifft, 2017-02-02:
With the new model pre-sets feature, the default RC expo should go to something more useful for Aeroplanes ie 25% rather than the Quad 70%

DzikuVx, 2017-02-02:
Please make a ticket in Configurator project

stronnag, 2017-02-04:
**Build Tested** : 50a53b76
**Hardware**:
SPRF3 on 260mm dead cat, M8N, BMP280.
SPEVO, 600mm tricopter, M8N

**Modes Tested**:
Free flight (horizon, LOS)
WP, PH, RTH, Auto land with disarm

**Special Points**:
Configured via CLI diff all
Default (racer) notch settings from latest configurator.

**Results**:
Just Perfect!

stronnag, 2017-02-05:
**Build Tested** : dcc163f185486d46b00f6ef6b5b66814dd444158 (alpha-2)
**Hardware**: 
* SPRF3 on 420mm quadt, M8N, BMP280.

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (450mm class) notch settings from latest configurator.

**Results**:
* Perfect 
* No regressions or bad behaviours noted.

flounderscore, 2017-02-06:
**Build Tested** : 73a5459 (Pre-alpha)

**Hardware:**
- Omnibus F3 w/ SD and Baro on heavy 26" flying wing
- M8N, no compass

**Modes Tested:**
- Acro

**Special Points:**
- Very windy and gusty
- D=0 for RP

**Results:**
- No regressions in flight
- Altitude display on OSD blinking
- PID displayed on OSD (has been fixed in newer revision, I believe)
- Logging to SD card worked. (No surprise, I guess, but a first for me)

stronnag, 2017-02-10:
**Build Tested** : c305dba225bd6872ac1b48c55681379db0bcf764 merged with PR https://github.com/iNavFlight/inav/pull/1250
**Hardware**: 
* SPRF3 on 260mm deadcat, M8N, BMP280.

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (250mm class) notch settings from latest configurator.
* Spektrum pseudo-RSSI PR (nice)
* Save / Restore WP to EEPROM

**Results**:
* Perfect 
* No regressions or bad behaviours noted.
* Mission loaded to EEPROM the previous day. Single key in the field from CGS (mwp) to restore from EEPROM / populate CGS display (and validate). Very convenient.
* Spektrum pseudo-RSSI / fade count working as expected.

digitalentity, 2017-02-11:
@stronnag thanks for testing! Mission Save/Restore is a very cool feature that was wanted for quite some time and after a year of development we got it working! :beers:

giacomo892, 2017-02-11:
Build Tested : 1.6 Alpha 2

Hardware:
Maple Mini (NAZE) on 480 custom built H wooden quadcopter with  BMP280 and MPU6050

Modes Tested:

Free flight (ANGLE);
Nav ALT HOLD

Special Points:

Configured via CLI diff all
PPM INPUT

No regressions or bad behaviours noted.

digitalentity, 2017-02-11:
**Build Tested** : 160e5ac991d927acee29dbaf3a09e6f724d4e81a
**Hardware**: 
* CC3D on Volantex Ranger 757-4 1.4m airplane

Modes Tested:
* Free flight (ANGLE)
* Auto launch (NAV LAUNCH)

**Special Points**:
* Configured via CLI `diff all`

**Results**:
* No regressions or bad behaviour

stronnag, 2017-02-12:
**Build Tested** : 490c164baf26e8221856f4da11ca3f2c72715eb6
**Hardware**: 
* SPRF3 on 420mm quad, M8N, BMP280.
* SPEVO on 600mm tricopter, M8N

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (450mm class) notch settings from latest configurator.
* Spektrum pseudo-RSSI PR (nice)
* Save / Restore WP to EEPROM (nicer:)

**Results**:
* Perfect 
* No regressions or bad behaviours noted.
* Mission loaded to EEPROM prior to leaving home. Single key in the field from CGS (mwp) to restore from EEPROM / populate CGS display (and validate). Very convenient.
* Spektrum pseudo-RSSI / fade count working as expected. 

Also tested arm safety modes with WP1 < 100m and > 100m from home. Working as expected.

Dronek, 2017-02-13:
Build Tested :  dcc163f
Hardware:
•SPRF3 deluxe (w/ active baro and mag) on FPV49 (130cm flying wing), M8N.

Modes Tested:
•Free flight (ACRO)
•LAUNCH, ANGLE+MAG, ALTHOLD, ALTHOLD+POSHOLD, RTH

Special Points:
•increased P(pitch: 40, roll 60) due to 30% servo rate
•FF=0 
•CRSF protocol for steering and telemetry

Results:
•acceptable stability on a windy day (no tuning done)
•[CRSF telemetry issues](https://github.com/iNavFlight/inav/issues/1285)
•[video and CLI dump](https://www.rcgroups.com/forums/showpost.php?p=36859809&postcount=11103)

digitalentity, 2017-02-13:
Build tested: b710cedc27be2e305dd13e589951020169305067

**Hardware**: 
* OMNIBUS on 800mm S800 wing

Modes Tested:
* Free flight (ANGLE, FPV)

**Results**:
* PIFF performance - perfect
* Some attitude drift noticed - https://github.com/iNavFlight/inav/pull/1284 might be the solution (test pending)

stronnag, 2017-02-16:
**Build Tested** : # INAV/SPRACINGF3 1.6.0 Feb 15 2017 / 16:52:52 (1c373bf6) [RC1]
**Hardware**: 
* SPRF3 on 260mm deadcat, M8N, BMP280.

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (250mm class) notch settings from latest configurator.
* Spektrum pseudo-RSSI PR (nice)
* Save / Restore WP to EEPROM
* black box denom of 50

**Results**:
* No flight / nav regressions (really nice flight characteristics)
* Mission loaded to EEPROM the previous day. Single key in the field from CGS (mwp) to restore from EEPROM / populate CGS display (and validate). Very convenient.
* Spektrum pseudo-RSSI / fade count working as expected. 
* Strange blackbox time stamps:
````
loopIteration	 time (s)
0	35.907236
50	35.957814
64	35.972139
114	36.022701
128	36.036887
178	36.0875
192	36.101595
242	36.152211
256	36.166564
306	36.217181
320	36.231354
370	36.281968
384	36.296125
434	36.346723
448	36.361064
498	36.411939
512	36.426045
562	36.476647
576	36.49087
626	36.541429
640	36.555556
690	36.606165
704	36.620646
754	36.671079
768	36.685231
818	36.735792
832	36.749916
882	36.800517
896	36.814751
946	36.865358
960	36.879465
1010	36.930006
````
i.e. alternating loop iteration intervals of 50 and 14, not quite what I expected: or coincidence that 50+14 = 64, a multiple of 32?

stronnag, 2017-02-25:
**Build Tested** : bc6d74d9990eaf95266f20b3efe144206bda83d0
**Hardware**: 
* SPRF3 on 420mm quad, M8N, BMP280.

**Modes Tested:**
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (450mm class) notch settings from latest configurator.
* Save / Restore WP to EEPROM 

**Results**:
* Perfect  (17m/s /  33 knots gusty wind); fun to watch how much it gets blown around changing heading at WPs
* No regressions or bad behaviours noted.

sppnk, 2017-02-25:
Build Tested : dcc163f
Hardware:

CC3D on MINITALON, AFRONAZE on miniSkywalker.

Modes Tested:

Free flight (ANGLE).
ALTHOLD, FAILSAFE (RTH)

Special points:

LTM telemetry working perfect in EZGUI at 9600 bps through openlrngs modules. First time it works so well. FPV flights with minmosd in the same port and LTM. All working perfect.

Results:

No problems with miniskywalwer, but minitalon doesnt launch so well (some times to the ground, ,missconfigured picth parameters?)

oleost, 2017-02-26:
**Build Tested** : RC2
**Hardware**: 
* Airbot F4 mpu6000 on 4s 210mm quad

**Modes Tested:**
* Free flight (Acro)

**Special Points**:
* 5 inch preset, with change to looptime to 4kHz. ( Around 250 Cycle time )

**Results**:
* Perfect, one of the best tuned quad I have flown.

`# status
System Uptime: 300 seconds
Voltage: 9 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, SONAR=NONE, GPS=NONE
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 3131, max available config: 131072
System load: 6, cycle time: 263, PID rate: 3802, RX rate: 49, System rate: 10

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       3       1    0.5%    0.5%         3
 1 -          PID    3921      78      44   31.0%   17.7%     54335
 2 -         GYRO    1956     113      16   22.6%    3.6%      9444
 3 -          ACC     494      20      12    1.4%    1.0%      1842
 4 -     ATTITUDE      99      23      15    0.7%    0.6%       473
 5 -           RX      49      47      37    0.7%    0.6%       575
 6 -       SERIAL      99     250       2    2.9%    0.5%       138
 7 -      BATTERY      49       7       2    0.5%    0.5%        32
 8 -       BEEPER     100      10       2    0.6%    0.5%        76
13 -    TELEMETRY     247      12       3    0.7%    0.5%       278
16 -          CMS      59       8       2    0.5%    0.5%        31
Task check function             6       1                        27
Total (excluding SERIAL)                    59.2%   26.0%

# diff

# version
# INAV/AIRBOTF4 1.6.0 Feb 20 2017 / 10:46:30 (7de1b48)

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 38400 0 115200
serial 30 32 115200 38400 57600 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 1 2 1300 1700
aux 2 2 2 1675 2100
aux 3 20 0 1300 2100

# adjrange

# rxrange

# master
set looptime = 250
set gyro_sync = ON
set gyro_sync_denom = 4
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 329
set acczero_y = -50
set acczero_z = -623
set accgain_x = 4058
set accgain_y = 4087
set accgain_z = 4010
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set min_throttle = 1030
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set small_angle = 180
set nav_extra_arming_safety = OFF
set telemetry_inversion = ON
set attitude_task_frequency = 100
set async_mode = ALL

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 65

# `

stronnag, 2017-02-26:
**Build Tested** : 178d105bbf865f3af62119301c81e1fd842b140b
**Hardware**:  * SPRF3 on 260mm deadcat, M8N, BMP280.

Modes Tested:
* Free flight (horizon, LOS)
* WP, PH, RTH, auto land / poweroff

**Special Points**:
* Configured via CLI `diff all`
* Default (250mm class) notch settings from latest configurator.
* Spektrum pseudo-RSSI PR (nice)
* Save / Restore WP to EEPROM

**Results**:
* Stupidly windy. Flying windward / leeward (upwind / downwind) missions, interesting to see that the vbat was reported 0.5v lower on the upwind legs. Not often that vbat goes up with flight duration.  Just like sailing in lots of breeze; plan for the downwind turn early.
* But perfect as usual. Copes wonderfully with lots of breeze.

sppnk, 2017-02-27:
Build Tested : dcc163f
Hardware:

CC3D on MINITALON, AFRONAZE on miniSkywalker.

Modes Tested:

Free flight (ANGLE).
ALTHOLD, FAILSAFE (RTH), FAILSAFE by switch and by kink lost.

Special points:

None.

Results:

Minitalon autolaunching nicely on 20 degrees climb. Not so well when activating failsafe by switching off the transmitter. The plane goes full throttle and weird trajectory. Blackbox doenst work in  my cc3d setup.

closed 2017-02-28 by digitalentity