iNavFlight/inav #1181: Unable to arm fixed-wing build on iNav 1.6

oleost, 2017-01-28:
[image]

Arm mode doesnt not light up.
[image]

[image]

Servo controll is working, and when using `Motor` tab and manually applying throttle to motor the motor spins.

Status, bootlog and dump in pastebin. http://pastebin.com/rVDdErWW

flounderscore, 2017-01-28:
I’m having the same issues. Flashed OMNIBUS from the latest development. But in my case, it’s possible I’m missing something, although all checks show up green the the configurator and I have GPS lock.

digitalentity, 2017-01-29:
Interesting. I've just tried on CC3D w/o GPS and it armed without issues. I'll find a GPS module and test again.

digitalentity, 2017-01-29:
@oleost I believe this is caused by #1182 (easily verifiable by trying arming by sticks). Fixed by #1183

oleost, 2017-01-29:
Closing this one as the new created issue will servo better. #1182 

( Did try with stick arming, first time also, but failed to realise that I didnt have setup my yaw channel on my taranis on the wing 👍  )

closed 2017-01-29 by oleost

cimenx, 2017-04-24:
I have the same problem with sp racing f3 in flying wing setup

digitalentity, 2017-04-24:
@cimnex please verify that you have all pre-arming checks on main configurator screen all "green" and stick mapping correct.