iNavFlight/inav #1182: Switch arming doesn't work

digitalentity, 2017-01-29:
Looks like switch arming sometimes is not detected. Might be the root cause for #1181 

Steps to reproduce:

1. Clean the board config
1. Configure remote
1. Calibrate accel
1. Verify that stick arming works
1. Configure ARM mode
1. Verify that switch arming works and stick arming doens't work
1. Reboot the board
1. Switch arming doesn't work BUT stick arming works regardless of ARM mode.

digitalentity, 2017-01-29:
Looks like `useRcControlsConfig()` is not called on boot which causes `isUsingSticksToArm` to stay `false` on boot regardless of mode config. @martinbudden I believe this was a side-effect of PG conversion.

digitalentity, 2017-01-29:
We probably are missing a call to `useRcControlsConfig()` from `activateConfig()`

martinbudden, 2017-01-29:
Yes, certainly seems an error introduced by the conversion.

I think we should rename `useRcControlsConfig` since it is no longer used to set the old config pointers.

oleost, 2017-01-29:
Bench tested, can confirm fix #1183 working.

flounderscore, 2017-01-29:
Same here. Bench test went fine.

digitalentity, 2017-02-07:
@martinbudden can you remind me why you reopened this?

martinbudden, 2017-02-07:
I think it was so that we would remember to rename `useRcControlsConfig`, since that name is no longer appropriate.

martinbudden, 2017-02-08:
Addressed by PR https://github.com/iNavFlight/inav/pull/1251

closed 2017-02-08 by digitalentity