iNavFlight/inav #1190: Flysky IBUS telemetry not work properly

hali9, 2017-01-29:
Data on the receiver display for a fraction of a second during startup, then do not.
I corrected as beelow and compiled. Now is good.
Inav: 1.4 (file ibus.c in 1.4 and 1.5.1 is the same)
Target: SPRACINGF3
TX/RX: Flysky-i6/Flysky-IA6B
Ibus.c:
#include "io/serial.h"
#include "fc/rc_controls.h"
**#include "scheduler/scheduler.h"**

**#define IBUS_TASK_PERIOD_US (500)**
#define IBUS_UART_MODE     (MODE_RXTX)
#define IBUS_BAUDRATE      (115200)
#define IBUS_CYCLE_TIME_MS (8)

void checkIbusTelemetryState(void) {
    bool newTelemetryEnabledValue = telemetryDetermineEnabledState(ibusPortSharing);
    if (newTelemetryEnabledValue == ibusTelemetryEnabled) {
        return;
    }
    if (newTelemetryEnabledValue) {
        **rescheduleTask(TASK_TELEMETRY, IBUS_TASK_PERIOD_US);**
        configureIbusTelemetryPort();
    } else {
        freeIbusTelemetryPort();
    }
}

DzikuVx, 2017-01-29:
@hali9 could you please make a proper Pull Request from this code? I would really appreciate it

hali9, 2017-01-29:
Done

oleost, 2017-01-29:
#1191

DzikuVx, 2017-02-02:
Since implementation in place, I'm closing it. It's scheduled for 1.6
@hali9 you might want to confirm with latest `development` branch

closed 2017-02-02 by DzikuVx