iNavFlight/inav #1192: Adding Distance to Home in Smartport Telemetry

elepete, 2017-01-29:
In case of interest for others.
To measure distances I am sending (uint32_t)GPS_distanceToHome in the RPM field.
With that I can measure distances to my (home location). To measure distance between 2 positions away from home location I am reseting home location by a switch during flight.
The code change needed in smartport.c is simple, and can be posted in case of interest.
Just posting here in case this could be of interest as feature to others.

DzikuVx, 2017-01-29:
Why not to add it as a custom sensor? SmartPort protocol allows it after all?

rb1205, 2017-01-30:
If that's the case then the best way would be IMHO to include the home coordinates. It's trivial to get the distance from it and you can use it for more advanced stuff, like heading-to-home or a map.

digitalentity, 2017-02-07:
Anyone going to implement this?

DzikuVx, 2017-02-07:
I might give it a try as Custom sensor. With my schedule, not sooner than march :)

digitalentity, 2017-02-07:
@DzikuVx so 1.7 is fine then :wink:

elepete, 2017-02-12:
@DzikuVx - thank you, but you may not need to do that
I have all code (which is tiny) misusing RPM ready - I will modify to a custom sensor and share, so no need for you to do that work - having that by end of March is no issue

DzikuVx, 2017-03-15:
Implementation in place in #1414

closed 2017-03-15 by DzikuVx