iNavFlight/inav #1199: LED Strip in 1.5.1 Naze...confused!

GBIAN, 2017-01-31:
I read the Getting Started...hard to believe, but I really did!

Looking to make LED strip work on Naze32 Rev 6.

Wiki says you need to enable in the code using cygwin.

I followed the instructions and looked at the target.h (Naze) for 1.5.1 and found that LED is already defined.

So in the GUI when I enable the option I show 0 available leds.

What did I do wrong?

GBIAN, 2017-01-31:
When I look in the CLI LED_Strip shows up as a feature that is enabled

GBIAN, 2017-01-31:
[image]

GBIAN, 2017-01-31:
[image]

oleost, 2017-01-31:
https://github.com/iNavFlight/inav/blob/master/src/main/target/NAZE/target.h#L188

Two slashes in front means that it's commented out and is basically not there.

GBIAN, 2017-02-01:
Oleost. Thank you. I'll try changing it to defined.

Steep learning curve happening over here!

GBIAN, 2017-02-01:
Did the edit and tried to compile.

Here is what I got;
 /cygdrive/c/dev/inav
$ make TARGET=NAZE
%% startup_stm32f10x_md_gcc.S
make: arm-none-eabi-gcc: Command not found
make: *** [Makefile:738: obj/main/NAZE/startup_stm32f10x_md_gcc.o] Error 127

What did I do wrong?

tonyintn, 2017-02-01:
In order to add back the LEDs you will need to give up something.  Example in my case I sacrificed Blackbox

GBIAN, 2017-02-02:
I did that and still got the same error message when I compiled.

[image]
$ make TARGET=NAZE
%% startup_stm32f10x_md_gcc.S
make: arm-none-eabi-gcc: Command not found
make: *** [Makefile:738: obj/main/NAZE/startup_stm32f10x_md_gcc.o] Error 127

oleost, 2017-02-02:
Your build environment is not set up correctly, either you forgot to install something or you have forgot to set up path. Follow the guide to the letter.

tonyintn, 2017-02-02:
****make: arm-none-eabi-gcc: Command not found****
Yep, I would say your path is incorrect the Environment variables must match your arm-none-XXXX path.

This might help keep in mind that vid is old so the path and version he is showing is no longer valid but the principle is the same.
also 
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md

GBIAN, 2017-02-02:
Thats what I was thinking it might be, I'll double check it all. Thanks.

digitalentity, 2017-02-07:
This issue seems to be support-related. Closing

closed 2017-02-07 by digitalentity

GBIAN, 2017-02-08:
Where do I continue to persue my support-related issue?