iNavFlight/inav #1209: Allow saving `SERVO AUTOTRIM` by disarm stick command itself

oleost, 2017-02-01:
When `fixed_wing_auto_arm` feature is used with `SERVO AUTOTRIM` you cannot save trim, simple solution is now temporarly turn off `fixed_wing_auto_arm` and do the trimming and then save using stick commands and enable it again afterwards.

But tried coding it, so it would save trim on the stick command itself, but I didnt manage to do it.

What I tried was changing:
https://github.com/iNavFlight/inav/blob/8d88d8056fa8dd88646b3d8c299705d846c1cb99/src/main/flight/servos.c#L492

`                if (!ARMING_FLAG(ARMED)) {`
to
`                if (!ARMING_FLAG(ARMED) || (rcSticks == THR_LO + YAW_LO + PIT_CE + ROL_CE)) {`

But this does not of course work since rcSticks is not a part of servos.c

Any other idea to make this simple work as described?

oleost, 2017-02-01:
If #1211 gets implemnted this could be easier to do?

digitalentity, 2017-04-25:
This is tricky. EEPROM write takes time. One can accidentally issue EEPROM write stick combo in flight and for duration of EEPROM write (quite long on F4 boards) FC will hang up.

digitalentity, 2017-04-25:
I'm moving this to 1.8 - we need to figure out a way to make this possible while keeping it safe.

digitalentity, 2018-02-25:
I'm still not sure it's safe to save this trim results by disarm stick command.

closed 2022-05-07 by DzikuVx