iNavFlight/inav #1215: GPS + HoTT Telemetrie with Revo F4 FC

Ralfde, 2017-02-01:
When I activate GPS on the main serial port, HoTT Telemetrie on the flexiport (4pin) and SUMD on UART 6, it does not work all together.
I use the Openpilot Revolution F4 (Revo) Board.

Without the HoTT telemetry it seem to work.

digitalentity, 2017-02-01:
Please define "doesn't work". Board doesn't boot, hangs up, something is
not functioning etc?

Ralfde, 2017-02-02:
SUMD on UART6 and GPS on UART3:  The prearmimng check shows hardware health fail and navigation is safe fail. GPS seem not to work correct.

SUMD on UART6 and no GPS, but HOTT Telemetry on UART3: Telemetry does work. The transmitter does show an Electric Air sensor. But no hardware health failure. But the vario values show wrong values.
-500 m and -1.2m/s. Please correct. The value 0 is -500 m. So please check for the offset. The value 120 = 0m/s.
The vario does also not react on pressure changes.

 SUMD on UART6 and no GPS, but HOTT Telemetry on UART1: hardware health pass, but HoTT Telemetry does not work. No Electric Air sensor found.

SUMD on UART6 and GPS on UART 1, but HOTT Telemetry on UART3: hardware health fail

SUMD on UART3 and GPS on UART1: hardware health fail

SUMD on UART1: hardware health pass

SUMD on UART1, no GPS, HoTT telemetry on UART6: hardware health pass, but no Electric Air sensor found. No telemetry.

HoTT Telemetry on UART6, SUMD on UART3, Telemetry does not work. Hardware health pass.

My GPS doesn't work any more. Need to test it, why. But HoTT telemetry does work only on UART3.

digitalentity, 2017-02-02:
Hm... Looks like your GPS module isn't functioning. I'm not a big expert in HoTT telemetry, there might be harware compatibility issues. Can you verify if HoTT telemetry works with i.e. Betaflight?

Ralfde, 2017-02-02:
The GPS problem was caused by a died GPS voltage regulator. But the HoTT Telemetry works only on UART3 and the vario data have the offset and do not show the correct altitude and m/s. Please correct this.

digitalentity, 2017-02-02:
So, to summarize:

 * HoTT telemetry works only on UART3 (Flex port)
 * HoTT telemetry shows incorrect altitude and climb rate data.

@Ralfde is this correct?

digitalentity, 2017-02-02:
@Ralfde HoTT in INAV only supports EAM sensor (battery status) and GPS sensor (coordinates, altitude above mean sea level, ground speed, distance and direction to home). There is no vario data reported via HoTT telemetry.

Ralfde, 2017-02-02:
Yes, but EAM and GPS include also vario datas, which should be used correctly.

digitalentity, 2017-02-02:
Since none of the devs have Graupner radio it might get tricky to develop and test this functionality fast. Scheduling for 1.7

Ralfde, 2017-02-02:
Dear Konstantin Sharlaimov,
would you or another programmer need a graupner radio for testing?Where should I sent it? 
To who should I send the HoTT protocol data?
Best regards,Ralf

digitalentity, 2017-02-02:
@Ralfde please email me at konstantin.sharlaimov@gmail.com, I'll handle better HoTT support. Thanks.

closed 2017-03-30 by digitalentity