iNavFlight/inav #1218: Battery voltage display with Openpilot Revolution (Revo) F4 board

Ralfde, 2017-02-02:
I added the voltage devider 10K to 2K2 like descriped here:
http://opwiki.readthedocs.io/en/latest/user_manual/revo/voltage_current.html

I get a voltage reading of only 0.5 V, instead of 12.3 V and I adjusted the voltage scale already to 200.
So the voltage reading is about factor 20 too low.
What is the reason for this? Is the voltage scale calcuated maybe wrong?

DzikuVx, 2017-02-02:
Is this the oroginal OP Revolution Board or some kind of a clone?

digitalentity, 2017-02-02:
If it's a cloned board it may already have a voltage divider onboard. Please consult your board manufacturer or vendor for this.

Ralfde, 2017-02-02:
It is the original one.

digitalentity, 2017-02-02:
For 2K2/10K divider (1:5.5 divider) you should use VBAT Scale value of 55. I've just tested and classic 1K/10K divider works fine on REVO

Ralfde, 2017-02-02:
Thanks a lot

digitalentity, 2017-02-07:
@Ralfde did you manage to get it working?

closed 2017-03-03 by digitalentity