iNavFlight/inav #1219: Time-based failsafe

digitalentity, 2017-02-02:
Limit amount of time pilot is allowed to fly and then initiate failsafe to land of RTH the machine.

Things to consider:

 * Preventing complete lock-out of RC control. Can be done by enabling time-limit as a box mode.
 * Don't enforce time-limit when close to home (based on GPS)

oleost, 2017-02-02:
Natural to look at voltage based failsafe at the same time?

martinbudden, 2017-02-02:
There are two reasons for time-based failsafe:

1. To prevent time overflow on F1 boards
2. As a generally useful feature.

For case (1) it should be enforced even when near home.

There should be two time limits. After time limit 1, the craft should execute the user's chosen failsafe behavior. After time limit 2, the craft should land. Time limit 2 should have an absolute max of about 1 hour 5 minutes on F1 boards.

digitalentity, 2017-02-03:
@oleost possibly.
@martinbudden what about allowing pilot's control when this kind of failsafe is triggered? I.e. at the moment airplanes can't land automatically which will cause crashes when this failsafe is triggered. We need something smart enough to be both robust and user-friendly.

martinbudden, 2017-02-03:
Yes, I see no reason pilots control should not be allowed. And for airplanes, I was referring to the "gently spiralling downward" mode which is sort of equivalent to landing.

One thing we certainly want to prevent is fly-aways on F1s when the system clock is about to overflow.

oleost, 2017-03-28:
#1475

Redshifft, 2017-03-28:
I Thought about this as a catch net - how to get quietly through the say 10 min loops evades me  atm :)

krzysztofmatula, 2017-04-05:
Isn't it possible to just workaround the F1 system time wrap in software? Hour-long flights may result in the aircraft being still far away from home if such case happens.

As a general concept - OK for triggering RTH, but rather not emergency landing if there's still enough power to continue RTH if time expires...

martinbudden, 2017-04-05:
> Isn't it possible to just workaround the F1 system time wrap in software?

Yes, of course it is possible. But it is a lot of work, could introduce bugs, requires a lot of testing and we already have a solution for F3 and above boards.

If you intend to fly long flights, then you should upgrade to an F3 or higher flight controller.

sppnk, 2017-04-06:
Still many people flies F1's. In RCG, Mini Talon plane long range pilots are discussing about this issue. My development skills are not enought to give a solution, so I am only reporting.

IMHO, solving the F1's timer overrun could extend the use of inav in all the rc airplane community.

Nowadays a CC3D is only 5$ in ebay, and almost the same for a VK2828U, so a full setup is amazingly affordable. The only two important issues for F1's are this timer issue and the failsafe/disarm one. With both solved INAV would fly much higher....

digitalentity, 2017-04-09:
Technically timer overflow shouldn't affect functionality, but it was never tested that everything will function properly beyond the 1h limit on F1. Also, F1 are at the limit of flash memory. Just a few more releases and NAZE and CC3D won't be supported any more.

teckel12, 2017-08-16:
I would suggest an auto RTH based on `batteryConfig()->batteryCapacity`  and `mAhDrawn` instead of time or voltage (or making the auto RTH function have an option to be based on the remaining battery capacity).  I do this currently and it's much more accurate than a time or voltage method.

A REALLY smart way to do auto RTH would be to take `mAhDrawn` and the time in the air along with `batteryConfig()->batteryCapacity` to calculate the mAh per second for the current flight.  Then, take the following variables:

```
set nav_auto_speed = 500
set nav_auto_climb_rate = 500
set nav_landing_speed = 300
set nav_land_slowdown_minalt = 300
set nav_land_slowdown_maxalt = 1500
set nav_rth_climb_first = ON
set nav_rth_alt_mode = AT_LEAST
set nav_rth_altitude = 2300
```

and calculate the time to RTH based upon current `GPS_distanceToHome` and `getEstimatedActualPosition(Z)` values.  That's basically what I'm doing but at the transmitter level currently via telemetry data.  I know what my variables are set as, so the calculations are mostly hard-coded.  But, at the flight controller level, this would be a really awesome way to create auto RTH.  I have a DJI Mavic and I don't even pay attention to the flight time, battery level, etc. because it will just auto RTH when it needs to.  Bringing this to iNAV seems like a solid idea.  Maybe I'll work on it.  At least sending an indication to the transmitter via telemetry that RTH is suggested based upon all the above logic.

fiam, 2017-10-07:
I think there's no point in implementing this as a failsafe trigger. Is better to have it as an alarm and let the pilot decide what to do.

While I do agree with @teckel12 that a "smart alarm" based on remaining capacity would be really nice, I think that is almost impossible to implement correctly on our case. We don't have a fixed power train, so consumption would need to be estimated and would have a lot of error. This means that, in order to allow users to rely on this feature we would need to trigger it very early in order to compensate for the error margin in power consumption. Not no mention that battery capacity would need to be manually configured and might not be correct since cheap batteries usually indicate more mah than they really have and usage will cause any battery to lose capacity.

TL;DR I think it would be nice to have but I fear that users will rely 100% on it causing their machines to drop off the sky before making it back because we can't make it as accurate as manufacturers which control the whole power train.

digitalentity, 2017-10-08:
Relevant issue #2199

digitalentity, 2017-10-08:
We're probably won't be implementing this. All currently supported targets have 64-bit time.

closed 2017-10-08 by digitalentity