iNavFlight/inav #1220: Connected GPS no Data

nightlightup, 2017-02-02:
I'm rather new to this INav software and everything it contains. Today I started my project and while trying to connect the Gps to a SPRacingF3 type FC I managed to connect everything the right way, I am getting the U-blox Neo-6M Gps connected but I am getting no signal from it. Sunny day outside everything connected and settings done the right way but no Fix even after 15 minutes. 
I am using a F3S AIO Pro FC - with the HMC5983L Compass onboard and a Micro HobbyKing GPS And Compass u-Blox NEO-6 and HMC5883. 
What could I have done wrong or what have I not done correctly ?

DzikuVx, 2017-02-02:
@nightlightup  Please share screenshot of following tabs in Configurator:

* Ports with visible GPS section
* Configuration with visible GPS section
* GPS with visible statistics section

nightlightup, 2017-02-02:
[image]
[image]
[image]
[image]

oleost, 2017-02-02:
Try setting

`inav_use_gps_velned = off
gps_dyn_model = pedestrian
`

Read https://github.com/iNavFlight/inav/blob/master/docs/Cli.md for what you need to know using these settings.

nightlightup, 2017-02-02:
i came upon an Issue with the fc, which is: it is supposed to have a HMC5983L Compass which I Nav  sees as: HMC5883. The same one is also on the Neo Gps Module. If i disconnect the gps compass is seen as 5883. If i connect the gps same compass is shown

oleost, 2017-02-02:
https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup

If you want to use the external magnetometer (built in in your GPS) and you have a FC with the same magnetometer (HMC5883L is very common), you have to disable it physically on your FC: remove chip from board or cut a trace. You can't use two identical chips/magnetometers on the same I2C bus.

oleost, 2017-02-02:
I really strongly advice reading through the wiki as it is made by the users of iNav and really covers what you NEED to know.

nightlightup, 2017-02-02:
If one would disconnect the SDA and SCA cables from Gps..shouldn't the conflict be resolved ???
if so then i am still not getting any data from GPS

nightlightup, 2017-02-02:
Upon first connection i had some cables arranged in a different way...when i connected the gps it did not receive power to the normal pin but to the rx pin. After rearranging them i had the power led on, so i presumed no harm was done but could it be the reason ?

andre92, 2017-02-02:
Yes, you can leave out the SDA and SCL cables from the gps and then you will be using the compass on the FC without any hardware conflicts. You just have to make sure it doesn't suffer from magnetic interference from the power lines. However, I do not see how this would cause your gps to not get a fix. The default cli settings for gps should work out of the box with Ublox gps. I only change gps_dyn_model to air_4g so I can do acrobatics. The weird thing is your gps tab indicates the gps is connected and sends 5 messages per second. The number of satellites is normally displayed as 0 when there is no fix.

If you actually burned out the RX pin on your gps, it is possible that the gps can still send messages but not receive its configuration from the FC, but I am not sure.

nightlightup, 2017-02-02:
Yes i knew about the TX and RX. The fact is that I Nav will tell you if they are switched because it will show a Red GPS icon on the upper right side. In my case if i take only the RX out it does the same ( red gps icon ) so i presume the RX is not dead yet... so I am still in the mist. Or maybe my stupid ass should buy another GPS as there could be a possibility that i have somehow fried the GPS. since I know nobody to use Inav and Ublox gps..around my area my only solution is to fork another 25$ for a neo 6m gps :(

andre92, 2017-02-02:
Did you try gpspassthrough with Ublox U-center yet?

DzikuVx, 2017-02-02:
From what I see, GPS->INAV connection works fine and no lock is because GPS module itself. My Neo-6M take sometime even 20 minutes to get a proper lock.

oleost, 2017-02-02:
@nightlightup Why not respond what I told you to try?

Also why would buy another neo 6m gps ? It's a none recommended GPS.

Go for an M8N, even better choose one of the mentioned in the Wiki which is known to work good, as some clones do not work any good.

nightlightup, 2017-02-02:
oleost....sorry. 
the two settings that you gave me did not make any changes....
I took out the sda and scl cables to eliminate the possible conflict.... still no go....
i went for the neo 6 because it was lying around and from what i have seen on youtube...there are not that many differences between the two ....8M can connect to Glonass while 6M works only with the US ones....

nightlightup, 2017-02-02:
little bugger works... no getting much info because my win pc is a tower and i can't mmove it close to the window

[image]

andre92, 2017-02-02:
Looks like your gps gets 3 satellites. iNav confirms gps lock with 4 or more, and will arm with 6 or more if you leave the arming safety feature enabled.
With less than 4, it might appear as if your gps isn't working in iNav. It is very useful to have telemetry on your transmitter, or use a blackbox to record gps performance (slowest BB logging rate will work, gps is only 5 hertz)

nightlightup, 2017-02-02:
thanks....i guess it is just a waiting game for the moment....will update when and if i get a fix for the moment...it just goes from 0-4 or so..but never getting a fix

DzikuVx, 2017-02-07:
OK, looks like it is solved, and GPS is sending data but never obtaining the 3D lock since. GPS module, not INAV issue. @nightlightup feel free to reopen if this is firmware problem after all

closed 2017-02-07 by DzikuVx