iNavFlight/inav #1221: pidsum_limit at 0 disabled pid for roll, pitch and yaw, is this intentional?

andre92, 2017-02-02:


andre92, 2017-02-02:
I was testing some preset modes in iNav 1.6 and suddenly the pid controllers stopped working. After some digging around in CLI I found pidsum_limit was set to 0, with allowed range of 100-1000. Sounds like a bug to me. Unfortunately I cannot remember what steps I took.

DzikuVx, 2017-02-02:
You need configurator 1.6

andre92, 2017-02-02:
I noticed my configurator has just auto updated. I cannot get the error back so maybe it was because of configurator. I also noticed 3d_deadband throttle at some value around 53k, must also have been a result of this.

closed 2017-02-02 by andre92

DzikuVx, 2017-02-02:
1. Open Configurator 1.6
1. Open PID Tuning Tab
1. Open Chrome developer console, for example by right click, inspect and go to Console
1. Type PID_ADVANCED
1. You should see `pidSumLimit: 0`
1. Type PID_ADVANCED.pidSumLimit=500`
1. Hit save in configurator

There is no GUI yet for this setting

andre92, 2017-02-02:
One more thing about presets, if I want to suggest changes, should I do it in inav directory or in the configurator's directory?

DzikuVx, 2017-02-02:
Configurator please. This is independent from firmware