iNavFlight/inav #1225: need different nav_rth_abort_threshold for FW vs MC

andre92, 2017-02-03:
Please consider using 2 seperate values for nav_rth_abort_threshold. The default of 500 meter may be too big for some multirotor pilots. I fly on a field that is not even that big and 500 meter would land my multirotor on a house or on a freeway. 50 meter would do for a MC.
I suggest splitting it in nav_MC_rth_abort_threshold and nav_FW_rth_abort_threshold

digitalentity, 2017-02-03:
You have presets. 
A lot of parameters are shared between multirotor and fixed wing but should have different values.

andre92, 2017-02-03:
Ok I will check out the presets and see if that will set a better value for multicopter.

digitalentity, 2017-02-03:
Currently it doesn't set different values for multicopters vs airplanes, but we should look into it.

digitalentity, 2017-02-03:
Let's continue the discussion in the Configurator.

closed 2017-02-03 by digitalentity