iNavFlight/inav #1226: DJI compass very slow to change

ShikOfTheRa, 2017-02-03:
I have two setups running DJI clone GPS/mag.

They work, however the compass is slow to react to changes. It takes seconds to respond to a 180 swing. Can watch as the heading data slowly changes until it approaches rest position.

I would guess maybe the updates are very slow in comparison to direct sensor read so filtering takes much longer?

I put the GPS/mag direct to OSD and the compass responds quickly

digitalentity, 2017-02-03:
Heading is calculated using gyroscope. Compass is only used to correct for drift. Are you sure you let the board calibrate gyros on power up (it takes a few seconds of complete stability).

Also, please verify that when you set mag_hardware=NONE your heading responds correctly to rotation.

If after all you observe compass drift when you rotate the board - play with mag_align setting - you most likely get it wrong.

sambas, 2017-02-03:
default orientation for NAZA GPS should be: #define MAG_GPS_ALIGN CW180_DEG_FLIP

ShikOfTheRa, 2017-02-03:
@digitalentity 
@sambas 
 Thanks guys for response

Of course -brain fart. I was just thinking the DJI puck was in default DJI orientation so no need to change, but that's not same as default iNAV. Makes complete sense now... doh

Thanks... consider closed. I know it will work when I amend later....!

closed 2017-02-03 by ShikOfTheRa

digitalentity, 2017-02-03:
@ShikOfTheRa yeah, flexibility comes at a price of increased complexity... Please let us know if you managed to get it working.

ShikOfTheRa, 2017-02-03:
@digitalentity 
Yes - works DJI compass now works good on the bench. Pending a fight test.

Thanks for your support...