iNavFlight/inav #1241: Issue with position of aircraft after about 1h and 30m

oleost, 2017-02-06:
Board: Spracingf3mini
Diff and full dump: http://pastebin.com/H5AJBjyc
Video of issue: https://www.youtube.com/watch?v=ZuonAJwootE

reboot of FC fixes the issue, tried leaving it on for 24h and then disconnect and reconnect, same issue appeared after 1hour+, so it can be temperatur related.

Screenshot of sensor tab, with special build from @digitalentity 

[image]
[image]

oleost, 2017-02-06:
Picture of sensor tab after when rebooted:

[image]

[image]

digitalentity, 2017-02-06:
This is a show-stopper for 1.6.

>dT jumps from positive to negative out of nowhere, one readout gets completely screwed and whole output is wrong
>happened after 1:15 more less
>so definetly it is connected with timer overflow

I confirmed the issue, it indeed looks like it's timer overflow. Possibly we forgot to convert to timeUs_t somewhere.

digitalentity, 2017-02-07:
Confirmed - issue happens when timer overflows over 2^32.

digitalentity, 2017-02-07:
Nailed it.

So is `micros()` returning gibberish when millisecond counter was over 1h15min (4294967 ms since boot).

In `micros()` we have this code:
`(ms * 1000) + (usTicks * 1000 - cycle_cnt) / usTicks`

While it looks ok, this part `(ms * 1000)` is yielding 32-bit result while is only converted to `timeUs_t` later when doing addition.

Adding explicit typecast fixed the isse:
`((timeUs_t)ms * 1000LL) + (usTicks * 1000LL - (timeUs_t)cycle_cnt) / usTicks`

PR pending

digitalentity, 2017-02-07:
https://github.com/iNavFlight/inav/pull/1243

martinbudden, 2017-02-07:
I've added test code to check this. Unfortunately the test code passes even in the old version - compiler differences I presume. Still the test code might pick up a future error. See PR https://github.com/iNavFlight/inav/pull/1244

digitalentity, 2017-02-07:
@martinbudden thanks. I presume different compilers do type conversion differently - `cc` may cast everything to final type and then do the math. Apparently `gcc-arm-none-eabi` does type-conversion differently.

digitalentity, 2017-02-07:
Closing. #1243 nailed it.

closed 2017-02-07 by digitalentity