iNavFlight/inav #1255: NONE failsafe procedure

rb1205, 2017-02-08:
As no flight is completely autonomous (not yet at least) there's always at least one phase where the human is controlling the uav. Conversely, the failsafe_procedure applies to all the flight from beginning to end, without considering if the uav is flying autonomously or is being piloted. Thus any "stable" failsafe condition happening outside of autonomous flight **will** result in a crash or a flyaway. This is why I think that the NONE failsafe procedure is intrinsically unsafe, and with no real uses in its proposed form. The best course of action IMHO is to always have a real failsafe procedure defined, then having a separate option (eg. failsafe_ignore_when_auto) that blocks failsafe from triggering while on autonomous flight.

rb1205, 2017-02-08:
This will also cover those cases (and I guess are the vast majority) where you use the same UAV both for autonomus missions and manual flights.

digitalentity, 2017-02-08:
Yes, NONE is intrinsically unsafe.

In next phase of failsafe rework we'll have failsafe procedures configurable for auto and non-auto modes. I.e. you can have RTL in manual and NONE in auto modes.

rb1205, 2017-02-08:
That'd be overdoing it IMHO. The only thing one should be able to configure is if the failsafe should trigger or not, but the failsafe procedure itself should be the same either way.

closed 2017-03-27 by digitalentity