iNavFlight/inav #1259: F1/F3 targets may cause interference on 433MHz LRS

digitalentity, 2017-02-09:
Quite a while ago (before going to airplanes) we removed `emf_avoidance` option. But F1 and F3 chips run at 72MHz, 6-th harmonic of which is 432MHz. This may cause issues with LRS operating in that band.

We need to think about a way to reduce interference to LRS to allow long-range airplanes to use INAV safely.

csurf, 2017-02-09:
Can emf_avoidance simply be reinstated? Why was it removed? What other options are there?

 I also need this for UHF setups on planes.

digitalentity, 2017-02-09:
`emf_avoidance` was overclocking the CPU thus violating the specs and making the system potentially less stable. We'll look into putting it back with possibility of _underclocking_ the CPU instead

martinbudden, 2017-02-09:
Yes, this should certainly be addressed by underclocking, not overclocking. Do we have any actual evidence of interference?

digitalentity, 2017-02-09:
No, not yet. I'll try collecting some using a spectrum analyzer.

oleost, 2017-02-09:
Why not allow slight overclock if people are using it with success?

My general impression is that these chips often can handle light overclock? 

Could we have some steps to take to ensure checking that chip will handle the overclock?

digitalentity, 2017-02-09:
Violating the specs is generally a bad idea. Manufacturers spec the chips to 72MHz for a reason - while 90% might work on 84MHz just fine, the remaining 10% might exhibit random unexplainable failures.

martinbudden, 2017-02-09:
It's one thing overclocking something sitting on your desk. Overclocking something that is flying through the air is not a good idea at all. Even if the failure rate is only 1% I don't think it is acceptable.

DzikuVx, 2017-02-09:
I really would like to know how realistic this scenario is. It's not like CPU or FC has something long enough to act as an antenna and radiate enough EM emissions to be able to interfere with something few centimeters from it (but more probably much further).

marbalon, 2017-02-09:
@DzikuVx like you know I made some boards in the past with F1 and RFM22B on the board and have no problems with long range fly - max was 53km but the limit was battery not RC link (and 1W power on Tx side). But that board using 16Mhz quartz if this change something. Now have new board also with embedded RFM22b as RC link but based on F3 + 8Mhz quartz and everything seems to work fine with no tricks. For now tested only on 23km with stable RSSI > 70%, more test when weather will be better. 

So I think it is not a big problem because this problem is mostly related with planes where you can put RX far away from FC.

DzikuVx, 2017-02-09:
Well, this is my point. I'm really not sure we should care about that

digitalentity, 2017-02-10:
Ok, I took an OpenLRSng module and used it as a spektrum analyzer. 
This is the picture of ambient noise in my room when only PC and OpenLRS module are powered up:
[image]

And this is the picture of ambient noise when I have a powered-up CC3D placed 5cm(!) away from OpenLRS antenna:
[image]

There is a slight increase of noise level near 432MHz (precisely 6th harmonic of 72MHz), but that's negligible. 

Here's the spectrum of ambient noise when I have OpenLRS module on the wing of my Z84 (AirHeroF3-controlled, powered-up and ready to go):
[image]

Just for comparison - this is the spectrum of 433.92MHz remote garage door opener operating in the next room (hardly a few milliwat power, 10m distance and a brick wall between remote and OpenLRS):
[image]

digitalentity, 2017-02-10:
Bottomline - FC hardly has any effect on 433MHz spectrum. Definitely not worth the effort implementing the underclocking feature.

For those who fear the RF noise - It's always best practice to put RC receiver as far from other electronics as possible.

rodizio1, 2017-02-11:
Sorry for re-opening, guess I saw this too late. If implementing underclocking is too much effort, it would be nice if you could atleast consider re-adding that overclocking feature.

The screenshot shows that the flightcontrol affects the 433MHz spectrum. If you switch the display to dbm, you can see that the flight control generates a signal that looks like something around 18db (!) above the noise floor. (One "20 point step" in the RSSI graph is 10db). 18db above noisefloor means you have 100x the energy in the "air".

Even when compared to the quite-high noisefloor in your area around 433MHz, this looks still like something around 3-6db. 6db means reduction to half range. Or to put it differently, to get the same range with the flightcontrol running, you need 4x the transmit power on your LRS transmitter to get the same range as without.

oleost, 2017-02-11:
@rodizio1 What happens if you do the same test with overclock feature turned on ( Use and old version or Cleanflight )

rodizio1, 2017-02-11:
It behaves exactly as expected, the harmonic moves from 432Mhz to 504Mhz. Made these graphs with a DVB-T dongle with a cheap DVB-T antenna which is not really made for such low frequencies. So with a real 433Mhz antenna it would look even worse.

Cleanflight 1.13 on a Afroflight rev5 with emf_avoidance = OFF
[image]

With emf_avoidance = ON
[image]

oleost, 2017-02-11:
@digitalentity ?

digitalentity, 2017-02-12:
@rodizio1 a question re your test - how far your FC was from receiver antenna? Was this a realistic test where your receiver is placed at least 30-40cm away from your flight control? What was connected to your FC?

This harmonic thing is interesting. Clock is essentially a square wave which should only have odd harmonics, so we shouldn't be seing 6-th harmonic at all - meaning we are seing interference caused by something else. @rodizio1 can you repeat your test with iNav on the same board and in the same environment?

martinbudden, 2017-02-12:
@digitalentity , we should probably just implement the underclocking option. It's not difficult and will allow users to experiment.

digitalentity, 2017-02-12:
To keep native USB operational we'll have to underclock from 72MHz down to 48MHz. USB controller operates on 48MHz and there's only 1 or 1.5 divisor possible for USB clock. If this is acceptable - let's do it.

martinbudden, 2017-02-12:
Yes, let's do it. 48MHz clock should work for some setups, if not all.

digitalentity, 2017-02-12:
Ok, let's plan it towards 1.7 - the change will touch the system code which may affect the whole families of boards. I'll implement something while we go trough RC-cycle with 1.6

rodizio1, 2017-02-12:
> @rodizio1 a question re your test - how far your FC was from receiver antenna? Was this a realistic test where your receiver is placed at least 30-40cm away from your flight control?

About 5cm. I run small quads. Also did some further testing, I need to be around 1.5 meters away with that crappy DVB-T antenna and not-so-sensitive DVB-T stick until the signal completely disappears.

> What was connected to your FC? 

Only power. 4 AA Battery holder, no switching BEC.

>This harmonic thing is interesting. Clock is essentially a square wave which should only have odd harmonics, so we shouldn't be seing 6-th harmonic at all.

This is only theoretically true for a 100% perfect square signal, which doesn't exist in real-world circuits.

>Meaning we are seing interference caused by something else.

You mean, the signal on 432Mhz that showed up in both my and your graphs (us being in different countries probably more than 1000km away from each other and using completely different equipment for the measurements ...), the harmonic moving up to exactly the anticipated frequency right after I typed "save" in the CLI and the fact that those signals disappear as soon as I switch off the FC is "something else"?

>To keep native USB operational we'll have to underclock from 72MHz down to 48MHz. USB controller operates on 48MHz and there's only 1 or 1.5 divisor possible for USB clock. If this is acceptable - let's do it.

Don't these FCs all have CP2102s on them? Never really looked into the hardware design of these boards, so I could be wrong. But I can tell that the USB ports on my Afroflight rev5 and SPRacing F3 deluxe still work with the STM32 clocked to 84Mhz.

>Yes, let's do it. 48MHz clock should work for some setups, if not all.

48x9 is 432Mhz ...

digitalentity, 2017-02-13:
>This is only theoretically true for a 100% perfect square signal, which doesn't exist in real-world circuits.
> 48x9 is 432Mhz ...

For one - we aren't seing odd harmonics, so this shouldn't be visible at all. Two - 9-th harmonic is in case of square (or triangle) signal 8 times less powerful than 6-th (corresponds to ~3 times less distance to get it below noise floor)

>Don't these FCs all have CP2102s on them? Never really looked into the hardware design of these boards, so I could be wrong. But I can tell that the USB ports on my Afroflight rev5 and SPRacing F3 deluxe still work with the STM32 clocked to 84Mhz.

Modern FCs use native USB, clocked from CPU PLL. SPRacingF3 and Naze use CP2102 on them which has it's own clock generator for USB.

digitalentity, 2017-02-13:
I see several constraints for this option to be implemented in INAV:

 1. It should mitigate the problem of interference
 1. It shouldn't break existing functionality - native USB should remain functional as in many cases it's the only convenient way to connect to the FC
 1. It shouldn't violate the specs for ICs - no overclocking

martinbudden, 2017-02-13:
I agree with the constraints. I think implementing underclocking to 48MHz is potentially useful. It gives users something to try if they are experiencing interference, and causes no issues for people who don't use it.

digitalentity, 2017-02-13:
I'll look into implementing underclocking option for F1/F3 chips in 1.7

rodizio1, 2017-02-13:
>For one - we aren't seing odd harmonics, so this shouldn't be visible at all. Two - 9-th harmonic is in case of square (or triangle) signal 8 times less powerful than 6-th (corresponds to ~3 times less distance to get it below noise floor)

But shouldn't the odd ones be the ones with higher power according to theory? Anyway, ~3 times less distance before getting the signal below noisefloor would still be 50cm, quite a lot.

>Modern FCs use native USB, clocked from CPU PLL. SPRacingF3 and Naze use CP2102 on them which has it's own clock generator for USB.
Also, FYI, no LRS operates precisely on 432-433MHz frequency. They usually have several channels scatterd over several MHz band, so very narrow-band noise at 432MHz shouldn't do much damage (if any at all).

Here, the legal frequency range is 433.05 to 434.79Mhz and 10mW transmit power. I have to "fight" for every decibel if I want to run a LRS legally (I don't need high range, just want to free the 2.4G band for video).

>1. It should mitigate the problem of interference - move the interference to other frequency or attenuate it

48Mhz would only attenuate it, not move it away completely.

> 2. It shouldn't break existing functionality - native USB should remain functional as in many cases it's the only convenient way to connect to the FC

So people would enable the option, type "save" and then be basically "locked out" on FCs which use native USB? That is really a showstopper.

> 3. It shouldn't violate the specs for ICs - no overclocking to avoid random issues caused by increased chance or hardware malfunction

I think while it's of course generally better to not overclock stuff like flightcontrols, this is more an academic discussion. Often, clockspeeds are not really near the maximum, they're more determined by other aspects like power consumption, other on-chip peripherals (and possible multipliers/divisors, like with USB ;)), marketing and whatever else.
Since 84Mhz has been widely used for quite some time now within baseflight, cleanflight and betaflight, I'd say it might be the more stable option in real-life, as there is no risk of something being too slow suddenly. I found for example when testing with sending MSP R/C commands, that Cleanflight starts to hang when sending too many commands per second (I think it was something around 150 or 200...).

But whatever, No. 2 is quite a showstopper and most people probably run larger copters or planes than me and use higher power and a wider frequency range, so 48Mhz is the best option. Thanks for re-considering :) It could be that I start to ask questions on how to change frequency in the source myself sometime ;)

ABLomas, 2017-02-18:
More about a problem - flying with OLRS on small minicopters for long time.
Actually, on small (180mm or less) you will have very small distance between RX and FC and without overclocking very noticeable. I even did some rough testing on field smth like 2 years ago - no lost packets in OLRS over range with overlocking (inject QUAL into PPM stream) vs several lost packets on far edge (including one longer failure, when i had no link for second or so).
Was flying with emf_avoidance until previous autumn, when it was removed and "consumed" way over 20 controllers, never had issues with overclocking.
IMO best case would be both over and underclocking (smth like emf_avoidance=1 and emf_avoidance=2), "1" being underclocking so even ppl with ancient dump files would pick safe value. Add note to wiki that any failures with overclocking turned on needs to be re-created with it turned off, done.
Yes, on bigger plane where you can have 50cm+ separation is not issue. But on micro-quads (they still can go way over 4km distance!) you have only 2-3cm and interference is really noticeable.

digitalentity, 2017-02-18:
Anything else thatn 48MHz and 72MHz will render native USB not working and we'll get tons of issues like "I can't connect to my EVO board after enabling overclock".

Overclocking is a feature that: 
a) Is locking user out of connecting to the board without any option to recover except re-flashing with full erate
b) Potentially unsafe

Even if it solves (theoretically) the EMF issue on F1/F3 boards - it's unsafe and very user-unfriendly.

This issue on hold until I have the hardware (SDR 0-1.2GHz receiver) to verify that issue exists and explore possibilities of fixing it.

DzikuVx, 2017-03-02:
@digitalentity I've made some tests with RTL-SRD and here are some results. 

# Flip32 F1 was running on 72,5MHz:

## 20cm 
[image]

## 10cm
[image]

## 5cm
[image]

# SPRF3 clone, was running slighly lower clock

## 20cm
[image]

## 10cm
[image]

# 5cm
[image]

# 2cm
[image]

# Conclusions

So, there is something on the topic, and 6th harmonic is visible in 433MHz range in case of F1 and F3 running on 72MHz. I've also tested one F4, nothing in range :)

But, looks like interference level is highly board specific. I had only one F1 and one F3 so was unable to confirm that F3 ale less noisy or just my Flip32 is more noisy. Either way, on 20cm even noisy F1 was indistinguishable from background noise. I live very near medical facility and 433MHz band is extremely busy here.

digitalentity, 2017-03-02:
@DzikuVx thanks for testing. Interference from the board looks very narrow band - it should be fairly easy to configure OpenLRS to avoid busy frequencies.

DzikuVx, 2017-03-03:
@digitalentity Yes, those peaks were very narrow. I was expecting to see some secondary peaks shifted left and right but there was almost nothing.

Schalonsus, 2017-03-16:
But i have another setup with EzUHF and a Naze32 rev4 or rev5 where FC and receiver are spaced about 30cm and there i have no problems.... so separation should be enough to solve the issue....

Redshifft, 2017-03-17:
Separation is always the first go to with RFI, thanks for sharing :)

Schalonsus, 2017-08-01:
Now i recognized the same problem in another plane. I have flown it before with Storm OSD up to 20km with nearly 100% link quality and now with F1 i am not able to reach 10km because of failsafes.

fiveangle, 2017-08-01:
[Here](https://github.com/iNavFlight/inav/issues/1259#issuecomment-278840489) someone wrote:

> There is a slight increase of noise level near 432MHz (precisely 6th harmonic of 72MHz), but that's negligible.

We're definitely not looking at the same picture.  The top of the grey "RMS average" area is what matters, and the picture shows it increases noise floor from a relatively flat ~82dbm to a decidedly peaky ~98dbm, which is huge for LRS.  Transient noise is just a part of life when using the UHF spectrum, but a 100ms blast from clicking a garage door opener 10 blocks away compard to sticking a constant noise source right under the VTX that increases noise floor 10dbm are vastly differnent scenarios.  There is nothing "negligible" about it.

I don't run F1 now, but when I did, I ran emf_avoidance and saw similar improvement, confirmed by Hawkeye DTFUHF JR module OpenLRSng spectrum analyzer, ImmersionRC EZUHF S.A., and RFExplorer standalone S.A..

Running UHF is so difficult as it is due to randomly spread noise but adding a constant noise source centered right at your hopping center potentially mm away from the RX reduces range considerably.

Schalonsus, 2017-08-15:
Yes, this is extremely disappointing.
My planes with UHF are long range planes or race wings i fly close to ground for few km, but without emf_avoidance its now impossible to fly the way i like. Seems i have to buy F4 boards or switch to mini apm... not really the way i like iNav.....

digitalentity, 2017-08-15:
F1 boards are not going to be supported by INAV for much longer. This leaves us with F3 boards. Due to most boards having native USB (without a separate USB/UART chip) we are limited to 72MHz or 48MHz CPU clock frequency otherwise we'll be unable to use USB port. Will 48MHz be ok for you?

Schalonsus, 2017-08-15:
I dont care about the further support of F1, its good as it is now with its features. In all my planes i have F1 except one F3. I would be really happy to be able to use 48MHz, i can disable a lot of feature on F1 i dont need.

digitalentity, 2017-08-15:
@Schalonsus it's a matter of implementation. 48/72Mhz will have to be implemented in system clock setup function - separately for F1 and F3. With ending F1 support I'm not sure it's feasible to implement it for F1.

Redshifft, 2017-08-15:
@digitalentity,  You may not have realized the little Afromini that many of us use (because it's small) does not even have a USB, I see no reason just to put that overclock option back in before you finish F1 support, won't fix problems for everyone sure but the option has been in there for years and I have not seen any problems reported.

fiveangle, 2017-08-15:
This debate is a bit silly. It worked perfectly for thousands (tens of thousands) and as mentioned, doesn't violate anything for FCs without USB (Afromini, every Flip32 with the microusb ripped off ;) etc)

Not implementing it does however make 433 UHF near useless for actual long range on these controllers.  If you had to pick your poison, the answer seems obvious [shrug]

digitalentity, 2017-08-16:
There is nothing silly about removing support for targets that are incapable of holding all features due to insufficient memory (Afromini, Flip32 - they all are F1s). In fact NAZE target already won't be around in next release. Therefore I don't think I'll be wasting time implementing underclocking for targets that are not going to be part of official release anyway.

F3 targets, however, are a different story - we'll implement underclocking to 48MHz to keep native USB operational and reduce effects on 433MHz band.

digitalentity, 2017-08-16:
See #2003 for implementation for F3 targets. Tested on SPRF3 and COLIBRI_RACE - seems to function properly. If anybody is interested in doing a similar implementation for F1 targets - feel free to submit a PR into `de_underclock_f3` branch before it's merged.

DzikuVx, 2017-09-08:
With #2003 merged, we can close this one

closed 2017-09-08 by DzikuVx