iNavFlight/inav #1260: msp on uart1

killerm1, 2017-02-09:
Hi,

I have an omnibus f3 with a broken usb port but have been able to run betaflight using uart1 and a ftdi. 
when I flashed inav I can no longer get a connection. Im assuming that msp is not enabled on uart1 by default being omnibus has a vcp. Am I correct in thinking this? and if so is it possible to enable it without a usb connection?

digitalentity, 2017-02-09:
Good point. Maybe we should enable MSP on UART1 regardless of VCP presence.

oleost, 2017-02-09:
I also have a friend that have exactly this issue.

What I need to change to let him have MSP on UART1?

killerm1, 2017-02-09:
I think it should be. you can always disable it if you dont require it on UART1.

digitalentity, 2017-02-09:
@oleost @killerm1 on it.

killerm1, 2017-02-09:
@digitalentity thankyou 👍

digitalentity, 2017-02-09:
I figure this was a side-effect of migrating to parameter groups config storage. Should be easy to fix.

closed 2017-02-09 by digitalentity