iNavFlight/inav #1262: Feature request, betaflight peripherals. Smartaudio and tramp vtx

oleost, 2017-02-09:
This is to control example immersionrc tramp vtx from FC( extended to example control vtx from Taranis )

https://github.com/betaflight/betaflight/pull/2106
https://github.com/betaflight/betaflight/pull/1283

https://github.com/betaflight/betaflight-configurator/pull/355

Example of use: http://www.nitbeatfpv.com/tramphv-vtx-settings-taranis

StewLG, 2017-03-15:
I'm interested enough in this to attempt the merge. Should I attempt a PR, or is there more formality/process in the way the project pulls from CleanFlight/BetaFlight?

tgreer, 2017-04-19:
Following as would like this too. @StewLG don't see any reason why you couldnt do a PR...

fiam, 2017-09-15:
Isn't this one fixed by #2084?

digitalentity, 2017-09-15:
Yes, looks like it. Closing

closed 2017-09-15 by digitalentity