iNavFlight/inav #1263: Feature request, MSP over smartport

oleost, 2017-02-09:
From betaflight https://github.com/betaflight/betaflight/pull/1438

Enabled example PID tuning from Taranis. http://www.nitbeatfpv.com/frsky-taranis-pid-settings

Also if peripherals is added it gives control of example tramp vtx http://www.nitbeatfpv.com/tramphv-vtx-settings-taranis

miradarya, 2017-02-13:
Do you think it could be possible to also use this as a link between UAV and GCS? Probably no way to connect the MSP serial port to a GCS on the Taranis so far? If this was possible, there would be no need for an additional telemetry link...

MichaelCWarren, 2017-02-27:
I also, would love this feature.

teckel12, 2017-10-02:
Actually, as part of my SmartPort (S.Port) telemetry refactor/enhancements #2050 iNav 1.7.3 already supports MSP.  Matter of fact, the default BetaFlight PID Lua script is totally functional with iNav 1.7.3.  There's some tweaks to be made with the Lua script for iNav, but it's totally functional (tried it yesterday with a Omnibus F4 Pro with telemetry over soft serial).

I'm working on a iNav fork of the Betaflight PID Lua script now.

teckel12, 2017-10-02:
@digitalentity @oleost see my comment above.

oleost, 2017-10-02:
Great, will test as soon as I get time.

I’m closing this issue, if it doesn’t work I will reopen.

closed 2017-10-02 by oleost

putimir, 2017-10-24:
@teckel12 can you give a newb like me some pointers on how make a setup like you described (quadcopter, I'm planning to use 1.8RC, Omnibus F4 Pro v2, with TBS micro v2 RX, GPS&Compass, Unify Pro HV). As a newb, I got lost when I used up my HW UARTs, and I'm trying to setup softserials; am also not sure if I need resource re-maping / or even know if that is possible in current version of iNav. Thank you in advance!

Jure

teckel12, 2017-10-24:
@putimir you have too many things you want to connect. Not enough ports. I have just GPS, compass & telemetry and I've used all UARTS and the soft serial port on the Omnibus F4. You have 3 ports to work with. Either 2 UARTS+SS or 1 UARTS+SS and one I2C. If you want GPS/compass, that uses a UART and I2C. So, after just the GPS, you only have the soft serial port left. You need to decide which ONE additional thing you want to add.

putimir, 2017-10-24:
Hm, I know this is not the right place to talk about this, but, if you allow me one more question; i can't get my head around this: what is then the purpose of "realocating / releasing unused resources": https://github.com/betaflight/betaflight/wiki/Single-Wire-Software-Serial (I know it's a BF, but the principles are the same right?) - you're supposedto be able to use PWM resources for either Tramp, S.port and S.audio...?!

putimir, 2017-10-24:
Another reference: https://www.rcgroups.com/forums/showthread.php?2809889-BF-Configurable-software-serial-ports

teckel12, 2017-10-24:
You can't reassign pins in INAV. UART1 is my SBUS port, so that was assumed to be you Rx port. I'm not aware of a single wire serial protocol in INAV. I know there's a multi-device serial protocol, but I don't think anything uses it at the current time.