iNavFlight/inav #1264: BMP180 does not hold reading, goes back to Zero.

techboycr, 2017-02-10:
I am using a CC3D board using version 1.5.1 of iNAV.

But the problem now is that when I moved the sensor up will read the change but quickly drift down to back to zero.

Lets say that goes up 40 cm but then goes back to zero even when is at the same altitude.

The BMP180 is connected in the I2C bus together with a magnetometer HMC5883l, test it with two different units and the problem persist, also check them with and arduino and both of them are working correctly. In the iNAV configurator is set as BMP085, does not recognize if set as BMP280.

Any help is highly appreciated.

digitalentity, 2017-02-10:
For 1.5.1 the "Barometer" graph at sensor tab actually shows estimated altitude which is pulled to zero until you arm.

closed 2017-02-10 by digitalentity

techboycr, 2017-02-10:
Thanks, and sorry for the dumb question.