iNavFlight/inav #1265: ESC Calibration imcomplete, missing calibration in motor 2 and 3.

techboycr, 2017-02-10:
My apologies if this is not the right place for this.

My Setup is CC3d/iNAV 1.5.1 with 12A Emax Simon Series ESC. I followed the calibration process but cant get it to work properly 1 and 4 get to move after calibration but 2 and 3 are missing the process.

Process follow if:

1. master to full.
2. Connect battery, and wait for beeps music beeps.
3. master to zero
4. Wait for beeps.

But there are not responding properly. It was wroking right with OpenPilot, this is the first test with iNAV.

Thanks.

DzikuVx, 2017-02-10:
Try calibrating one ESC at a time. Sometimes ESC have slightly different timings.
Since this is not firmware issue, but support, I'm closing it

closed 2017-02-10 by DzikuVx

eziosoma, 2018-09-09:
I have the same problem with INAV v 2, Seriously Pro F3.
How techboycr resolved?