iNavFlight/inav #1272: FW - set default small_angle to large value for fixed wing

csurf, 2017-02-11:
not really a big deal, but it's kind of annoying having to remember to disable small_angle for fixed-wing in order to allow arming the plane in any position.

Perhaps we could have the variable automatically set to '180' whenever a fixed-wing mix is selected?

Dronek, 2017-02-18:
Wouldn't it be actually more reasonable to completely get rid of this arming restriction even for multirotors?

csurf, 2017-02-18:
fine by me, anyone else think so?

oleost, 2017-02-19:
No this should stay as is. Its a safety feature.
Its prevent people from the aircraft when its tilted.
If people want to get rid of it, its easily configurable, its nothing the firmware enforces on users.

Honestly this belongs in configurator github, with changes to preset.

Airplane in general should probably have something like 70 to allow arming on ground, in hands and rearming in air more easier.

Racing quads could probably go for 180, to allow to misuse arming when stuck in trees etc to get down.

Bigger GPS / Arial platforms should probably stay as is, who wants to arms something like that tilted 45, or even upside down?

Closing this, feel free to reopen if not agreeing.

Try and open pull request / issue towards inav configurator github to change presets.

Dronek, 2017-02-19:
While I understand that small_angle was originally intended to be a safety feature, I think at the same time that it has become a questionable carryover from old days and especially obsolete in iNav due to following reasons:
a) prevention from arming with miscalibrated or malfunctioning accelerometer hardware: this is most likely covered by iNav's calibration procedures, hardware checks and extra_arming_safety. 
b) prevention from unintentional arming: unintentional arming should be considered as equally dangerous regardless of the aircraft's present orientation. Safety measures like e.g. two required consecutive (identical) arming commands on first arm might therefore be a safer way to go.
c) prevention from re-arming a non-leveled aircraft in mid-air after unintentional disarming: this is most likely the biggest drawback of small_angle, especially for fixed wing. IMHO the pilot's (re-/dis-)arming  authority should be as high as possible once the aircraft is launched. With the overall complexity of a flight control firmware including navigation capabilities it is fair and reasonable to expect an appropriate arming-responsibility from the pilot.

oleost, 2017-02-19:
Maybe we should increase default value to 50 deg ?

Redshifft, 2017-02-19:
small_angle = 180 for Aeroplane as default, if nothing else it opens the option for Disarm/Arm in the Air
which for a cool headed pilot might save the Plane sometime. Regardless of the fact many arm Planes with the prop pointing away or down etc,etc
The safety concerns have zero merit, the pilot nor the FC  controls the motor the ESC does !
when powered up having respect for any high powered device is all people need to understand.

csurf, 2017-02-20:
The more I think about it, I agree that this option could probably be safely removed. It should be up to the pilot to manage the safety of the model & the motor, and not a simple feature that relies on the angle of the model. For planes, I think it's important not to put any artificial limitations on the ability to arm and disarm.

sppnk, 2017-02-20:
I have the same opinion as @csurf

fiam, 2017-09-17:
@digitalentity I'I leaning towards setting small angle to 180 in the FW presets. Thoughts?

digitalentity, 2017-09-18:
Agreed

fiam, 2017-09-18:
@digitalentity We don't have any MSP commands for setting `small_angle` nor `imuConfig_t`, do we? Should we introduce a command for the whole imuConfig_t or just for `small_angle`?

DzikuVx, 2017-09-21:
No, we do not have MSP for that. Probably it's time to add a new one with settings that we can not set via MSP

digitalentity, 2017-09-21:
I think it's time to start migrating from MSPv1 to MSPv2 - introducing new MSPv2 commands to read sets of parameters (maybe even whole PGs)

fiam, 2017-10-12:
Fixed by #2274 and https://github.com/iNavFlight/inav-configurator/pull/279

closed 2017-10-12 by fiam

TronskiFPV, 2018-08-28:
Am I misunderstanding how this was resolved? On 2.0 final release my fixed wing default for small_angle was 25.

fiam, 2018-08-28:
The fixed wing presets where changed to set small_angle to 180.