iNavFlight/inav #1276: "navigation is safe" is off for SPRacingF3+Ublox M8N GPS under 1.5.1 and 1.6Alpha2

Civ1, 2017-02-11:
Hi, I dont know is it bug or not
When I set Failsafe/RTH or Modes/NAV POSHOOLD or Modes/NAV RTH under SPRacingF3+Ublox M8N GPS with 1.5.1 and 1.6Alpha2 versions "navigation is safe" become red (OFF) and copter not armed.
Sometimes  "navigation is safe" become "green" under version 1.5.1 with Failsafe/RTH but never with Failsafe/RTH+any GPS Mode.
All time GPS see 7-11 satellites and 3D FIX is green.

Under version 1.4 - no problems

Thanks.

MarcoPTG, 2017-02-12:
I have the same hardware that you have, and no issues.
Please recheck your settings.

stronnag, 2017-02-12:
No bug, flown dozens of SPRF3 / M8N flights without problems on 1.5.* and pretty much all flyable iterations of development/1.6. See https://github.com/iNavFlight/inav/issues/1172

digitalentity, 2017-02-12:
@Civ1 If you try to activate WP, RTH or PH prior to arming - the extra safety feature will prevent you from arming the machine. This is the expected behavior.

closed 2017-02-12 by digitalentity

Civ1, 2017-02-12:
Hi, thanks for answers, I see  "navigation is safe"  is RED during setup in iNav configurator. And cannot make it green with any used GPS options, like RTH. Is it correct ?

stronnag, 2017-02-12:
If you do not have the TX powered on, nav-safe will be red.

Nexus1212, 2017-03-30:
Hello,
I have the same issue. I use the Inav Version 1.6.1 with a SPRacingF3 and a NZ mini GPS. 

In the pre arm setup "navigation is safe" is always red. In the tab GPS is for few seconds my GPS position, then I get a error message.

digitalentity, 2017-03-30:
Either position accuracy is not good enough or you have WP, POSHOLD or RTH modes already activated in modes tab