iNavFlight/inav #1277: boxids mode bit not being set for PASSTHRU?

csurf, 2017-02-12:
I'm having an issue with MWOSD not seeing when my plane is in passthru. It still shows it as 'GYRO'. I checked out the MWOSD code, and it looks pretty straight-forward. It simply checks to see if the bitfield is set for passthru within BOXIDS MSP command response.

Can someone please confirm if the mode flag is correctly being set for PASSTHRU in BOXID MSP? Does configurator read the mode settings via MSP?

I will quickly close this out if this is a non-issue...

oleost, 2017-02-12:
Box IDs in iNav have changed. mwosd needs to change it based on if iNav is defined.

Copy from another answer I gave:

Mwosd fault I believe. It needs to be aligned with https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_msp.c#L112 when you define that you are using iNav.

https://github.com/ShikOfTheRa/scarab-osd/blob/4232212e849c269aeb60e05af59e3cd57214a26f/MW_OSD/Serial.ino

( Should probably be an if define INAV. Use the box names an ID from iNav code. ) You could maybe do it yourself or ask devs of mwosd, providing the first link )

csurf, 2017-02-12:
I don't understand how this would affect the PASSTHRU mode, as the order of the modes/ID's is the same up until PASSTHRU (mode ID#12), so MWOSD should still be able to pick it up normally.

Am I missing something here?

csurf, 2017-02-12:
@oleost, @digitalentity,
can you guys recommend an easy way to test if the passthru bit is being set within MSP_STATUS command?

the following function is called, which should set the PASSTHRU bit field...
https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_msp.c#L343

can't think of an easy way to test this out without writing a custom MSP parser in order to process it.

ShikOfTheRa, 2017-02-13:
I'll do a mwosd debug version for you tomorrow to view what is sent. I sent bixid 12 to osd and it lit up ok.

ShikOfTheRa, 2017-02-15:
https://github.com/ShikOfTheRa/scarab-osd/issues/379

csurf, 2017-02-19:
closing, was an issue with MWOSD

closed 2017-02-19 by csurf