iNavFlight/inav #1279: Feature Request : Inav + FBL

ukmook, 2017-02-12:
I would like to add Inav to a Helicopter by using it in conjunction with a Flybarless Controller.

I want to pass the outputs to a helicopter with a flybarless system already fitted, this would remove any need to rewrite Inav for a heli with all the weird Helicopter Dynamics, Physics etc, 

I am trying to add passthrough mode to the multirotor mixer, and map the outputs to a single servo system. (passthrough gives full control back to the pilot)

Servo 1 = Aileron
Servo 2 = Elevator
Servo 3 = Throttle / Collective / 1000us = -10Degrees : 1500us = 0Degrees : 2000us = +10Degrees of blade Pitch. (these numbers are not set in stone.
Servo 4 = Yaw
Servo 5 = Throttle set at a fixed position for the ESC Governor i.e. 60% = 1900 RPM Head Speed / 0% ESC Off (auto landing / Takeoff) 

We could add a feature to inav called Heli + Fbl

I think this would benefit a lot of heli flyers as it would be an inexpensive way of adding full Navigation, waypoints, RTH etc, also the code supports so many boards from the most expensive to very cheap that i'm sure it would further the Inav project.

I have bought a PWM to Sbus converter, but at some time in the future to keep wiring to a minimum a serial RX signal out from the Flight Controller would be the way to go.

oleost, 2017-03-11:
@ukmook Any progress?

ukmook, 2017-03-12:
No afraid not, looking for someone to help me. I can build the hardware (including sparky2)  but i'm not a coder, when i get a chance i will look at it again.

danarrib, 2017-11-09:
I don't know much about FBL Helicopters, but I guess the FBL is a kind of Flight Controller, since it's intended to stabilize helicopter attitude. So INAV should NOT use any PID corrections, right?

To navigate, it should NOT mess with throttle (throttle is always the same value during flight, right?)
To gain altitude, just increase blades pitch and to lose altitude, just decrease blades pitch.

This seams not too hard to add... But it'll be very difficult to test, since RC helis are expensive and any mistake will totally destroy it.

thegreen4me, 2018-05-24:
is there actually any reason inav couldnt just be a flybarless controller?  I know nothing about helis, but wouldnt it just be a mixer that controls 4 servos and thats it?

closed 2018-08-06 by stale[bot]