iNavFlight/inav #1282: Fixedwing doesnt try and descend after RTH

oleost, 2017-02-12:
# INAV/SPRACINGF3MINI 1.6.0 Feb 12 2017 / 00:15:25 (490c164)
mixer FLYING_WING
set nav_rth_allow_landing = ON

Folder includes some videos, some logs, DIFF and complete DUMP. https://drive.google.com/open?id=0B-yHzNtOXxbEdERWRWZLOEo0ZVE

Log 129 and video 172539 is together. To sync up watch when I flip arm switch on video. ( Last switch )

I let it circle for long time without it dropping any altitude..

PS ignore the failed launch, totally my fault. ( One time failed to turn up throttle and one time forgot I had not set up arm switch as part of bench testing )

digitalentity, 2017-02-13:
Bug confirmed. The plane doesn't understand that home is reached. Fix pending.

digitalentity, 2017-02-13:
A fixed-wing will circle around the home waypoint (at `nav_fw_loiter_radius` radios) without reaching it (need to be closer than `nav_wp_radius`). We need to consider home reached for fixed-wing when we are doing circular loiter around it.

PR #1289 (2017-02-14): Fix RTH landing on airplanes
Fixes #1282
closed 2017-02-14 by digitalentity