iNavFlight/inav #1283: Feature request: option to force POSHOLD during "climb first" stage of RTH for fixed wing

Dronek, 2017-02-13:
As of now "nav_rth_climb_first = ON" lets a fixed wing fly leveled in RTH mode until preset altitude is reached before changing direction to head home. In my opinion this behavior can result in moving the aircraft even more out of RC range in a failsafe triggered RTH event and/or unnecessarily increasing the distance to home in a manually triggered RTH event. Therefore it might be desireble to force POSHOLD during "climb first" stage of RTH, at least optionally.

digitalentity, 2017-02-13:
The reason for `nav_rth_climb_first` option existance is that a plane might lose altitude during the turn and/or it might be flying in the fiords where it's desireable to climb first before turning to avoid hitting the cliffs.

Poshold will make exactly the same kind of turn as RTH would do thus imposing same threats. If you are willing to take the risk of doing a turn at current altitude - just set `nav_rth_climb_first = OFF` and you'll start turning home instantly climbing on the way.

Dronek, 2017-02-13:
> If you are willing to take the risk of doing a turn at current altitude - just set  nav_rth_climb_first = OFF  and you'll start turning home instantly climbing on the way.

I see your point. There is certainly a multitude of imaginable scenarios favoring one or the other RTH behavior. Obviously I missed the fact, that without "climb first" the plane will climb on the way back when "nav_rth_alt_mode" is set right.

closed 2017-02-18 by digitalentity