iNavFlight/inav #1287: Difference between get and dump on various parameteres

oleost, 2017-02-14:
Looks like there is difference values returned from `get` and `dump / diff`

oleost, 2017-02-14:
Seems to be mainly isolated to mc_ and fw_ spesific pids.

digitalentity, 2017-02-14:
@oleost I can't reproduce the issue

digitalentity, 2017-02-14:
Everything is working as it should on CC3D board. Can you please re-test?

oleost, 2017-02-14:
Out flying now, could you test on an F3 based?

digitalentity, 2017-02-14:
Tested on EVO - no issues.

oleost, 2017-02-14:
https://youtu.be/SSFVw8lKEgM

oleost, 2017-02-14:
This shows much.

oleost, 2017-02-14:
So, its seems like somehow I selected profile, probably due to some accidental stick commands or something? However I tried chaning it back by typing `profile 0` but it doesnt seems to work.

digitalentity, 2017-02-15:
Will try reproducing today, thanks.

digitalentity, 2017-02-15:
Confirmed. `profile` command in CLI won't switch the profile and previous profile is still being used.

digitalentity, 2017-02-15:
Ok, figured it out. `profile` commands indexes are incorrect. 'profile 0' is meaningless.

digitalentity, 2017-02-18:
@oleost is this still relevant? I believe I've fixed the 'profile` command issue - it now accepts numbers 1-2-3 for Profile 1, Profile 2 and Profile 3 and dump/diff provides consistent results.

oleost, 2017-02-18:
I believe I saw a commit for you fixing it.
Since you have tested it probably safe to assume it fixed.

I'm in work period now and limited time to test.

If I I get any issues again I will reopen issue.

closed 2017-02-18 by oleost