iNavFlight/inav #1294: RX PWM not working on CC3D

dbechth, 2017-02-15:
I am unable to get RX PWM to work correctly on the CC3D and have tried versions from 1.6 Alpha2 back to 1.4 and get the same result. Since this is a first time setup of this board my config is defaults with RX_PWM checked. I see that there is a closed issue and pull request #267 #277 that seems to loosely match the issue I'm having. 

The table below is showing what pin corresponds to the Chanel map on the receiver tab in iNav configurator.
3: Nothing
4: Nothing
5: Throttle
6: Nothing
7: Aux1
8: Aux2

digitalentity, 2017-02-15:
By default PPM is configured. I've just checked FlySky and FrSky PWM RX and it works perfectly fine with CC3D.

Please provide more information, preferrably the full config dump of your settings using latest 1.5.1 or 1.6-RC1 firmware

dbechth, 2017-02-15:
Thanks, will provide dump when I get home..

dbechth, 2017-02-16:
No need for a dump, I figured out that pins 3-6 need to have a valid PWM signal at all times in order for the other channels to work. What tripped me up was reading the [cc3d documentation](https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20CC3D.md) which says that pin 3 is not used. With all 6 PWM channels now connected I report no further issues and am ready to get this bird in the air!! Thanks for the response and for the testing on your end!

closed 2017-02-16 by dbechth