iNavFlight/inav #1295: Strip down F1-based targets

digitalentity, 2017-02-15:
F1 targets are at their limit. We need to think about stripping them down to keep them afloat for next few releases. Better code modularity will allow us to exclude more features (hopefully unused) from F1-targets.

DzikuVx, 2017-02-15:
I'd say "legacy" branch.... We've been pushing the limits for quite some time now. And there will always be someone who will want LED strip on F1 and so on

digitalentity, 2017-02-15:
I'm leaning towards "legacy" as well. Backport only bugfixes there - no new features.

DzikuVx, 2017-02-15:
Yes, exactly like that. With some cleanups _master_ branch would be so much simpler if we remove all ifdef NAZE and ifdef CC3D.
`pwm_mapping.c` might even be understandable again

martinbudden, 2017-02-15:
Well, more modularity is a good thing. But I'm against creating a separate branch for F1 - I think that will make it more difficult to manage.

There are actually very few #ifdef NAZE and #ifdef CC3D left in the code.

giacomo892, 2017-02-15:
What about creating custom builds to removed unwanted features that pick lot of space in the F1.
I'm talking about unused sensors (SPI or I2C), NRF24 and so on.. 
Also could be nice to create a Fixed and Moving wing version for NAZE.

Schalonsus, 2017-02-15:
I'm for "I'm leaning towards "legacy" as well. Backport only bugfixes there - no new features."
And please take BMP085 back again into F1 (Afromini rev0) or does it take too much space?
ok, probably everything needs to much space for F1 😋

sppnk, 2017-02-17:
All my planes (now seven) since this project started are flying with F1's (cc3d and afronazes). Some of them have flyied amazing long fpv flights with minimosd and ltm telemetry. It is easy to see that everybody is moving to F4+ boards (maybe including myself soon), but I dont see the reason to drop F1's. Just some crashes and two lost planes, but because of the pilot, not the flight controller hardware or software :-)

F1's simply would not have the new features of the future releases, but they could stay alive, for planes at least, without many more requirements (LTM + OSD is amazing in such cheap boards).

Anyway, the 1.6 version is working fine in naze and cc3d. For now I dont need anything else, althought for sure I will enjoy very much to test new version in F3+ or F4+ boards.

Thanks you all for your efforts.

oleost, 2017-02-17:
One important thing to remember is that even if F1 is dropped it's not like current release is going to stop working or worse.

At some point the threshold will be reached where removing features may actually make the old builds "better" than new ones.

Ofc if one can magically save flash it's great to keep support, but if it's a matter of always removing features I also believe at some point it will need stop. And if any bugfixes makes it back into "legacy" we are lucky.

DzikuVx, 2017-02-18:
@oleost I think this is something most users forget. Not releasing new version for F1 does not mean older versions will not work

csurf, 2017-02-19:
I also have a few planes built on F1 (naze rev5).  F1 is ok for fixed-wing, and might still be able to benefit from some of the newer nav features/improvements that might be released in the near future. A custom-build system or some other means to select which features to include would be great. 

Isn't about time to separate Fixed-wing and MR code into separate branches? It seems like it would make things flow a lot better in the long-run. All platforms could share the core code, but then each platform can have custom-tailored code based on necesities and most common setup.APM does this, why can't iNav?

digitalentity, 2017-02-19:
We've considered separating FW and MR code, but it doesn't really make much difference - platform-specific code is minimal and benefit will not be worth the effort.

csurf, 2017-02-19:
ok, understood.

how about a custom build system where user can add/remove/select features and then download a build for that feature set? I'm thinking of something similar to how OpenTX Companion allows the user to select which firmware options are desired & then download the image corresponding to that selection.

Could something like this be feasible?

sppnk, 2017-02-20:
Yes, but I guess the already existent '#ifdef' preprocessor statements are the way to do 'custom builds'.

Anyway, if F1's are going to be droppped, at least this should be done with a good rock-solid and stable release (as 1.6 seems to be). And let these binaries to be easily downloaded from the configurator or github, because as @digitalentity said, there are still a lot of F1's out there.

martinbudden, 2017-02-20:
There are no plans to drop F1s yet. This issue is just to recognise the fact that they will become increasingly difficult to support and we should plan for that.

Schalonsus, 2017-02-20:
I also think 1.6 stable would be the right release to stop improving F1 targets. Just one thing i dont like: 32bit time restricting flight time....

sppnk, 2017-02-21:
Yes, the 32bit restricting flight time is a really bad thing to drop F1s for ever with that issue.

martinbudden, 2017-02-21:
@sppnk , 64bit time is not something that has been dropped from F1s. They never had it. It has been added for F3 and F4 targets.

sppnk, 2017-02-21:
I know @martinbudden , what I meant is that leaving F1s with that restricted flight time (because a bug or no bug) is not a good thing (IMHO). 

It was explained to me that it is an inherent problem for 32bits targets, but I dont know if there is any way to solve it. I so, I could open an issue...:-)

csurf, 2017-02-21:
OT: where can I learn about the 32-bit time limit issue? I googled around but couldn't find anything. Doesn't the time simply roll-over as soon as the uint32 overflows (this is how it works in ATMega-land) ? If not, then wouldn't it be possible to manually roll it over prior to having it overflow? Just curious...

Redshifft, 2017-02-21:
@csurf  "On F1 the system time will overflow at 1 hour, 11 minutes, 35 seconds and may cause some systems to malfunction" - there won't be many troubled by this limitation :)
F1 should be supported for a little longer because there are 1000's out there, just drop BB and LTM support to stretch it's life.

martinbudden, 2017-02-21:
Our judgement is that the 1 hour 11 minutes flight time is not a problematic restriction for F1 boards. It's not ideal, but it won't affect most users.

One thing we have also discussed is adding a time-based failsafe that triggers after (say) a hour for F1 boards.

If you are planning on doing really long flights, then you should get a F3 or F4 based flight controller.

DzikuVx, 2017-02-21:
@Redshifft dropping BB and/or LTM is not a real solution to a problem. It would only make problem more visible, since some users will want BB or LTM instead on >1h flight time.
F1 boards will be alive as long as they will work. Not releasing new version  does not mean they stop working out of nowhere

giacomo892, 2017-02-22:
Since compiling is very simple, if someone can made a wiki guide on how to create a custom target and customize the build could be very great :)

Schalonsus, 2017-02-22:
Flight times over an hour are really no problem at all for many people nowadays. I think it is no good idea to ignore this issue.

oleost, 2017-02-22:
I'm sure if it's was an easy task that limitation would already be gone.

But my guess the need for flight time above 1h is probably less than 1% of users. And if they need to overcome it F3 based only costs 20$ these days.

The most important I believe is that people know of that limitation.

martinbudden, 2017-02-22:
@Schalonsus , we are not ignoring the issue. We added support for long flight times in the 1.6 release - previously it was not supported on any targets.

We are extremely ROM-constrained for F1 targets, so we have to choose carefully what features F1 targets support. Extended flight times is one of those features that we have chosen to exclude from F1 targets. We have been explicit about this, so users know this is the case.

csurf, 2017-02-22:
still not clear why this is even an issue. counter roll-over is a common issue that's been dealt with even since 8-bit multiwii days.

http://stackoverflow.com/questions/3095623/how-to-deal-with-a-wrapping-counter-in-embedded-c

IIRC, if the micros()- related math is done in terms of 'time deltas' instead of relying on absolute values, then there shouldn't be any issues. 

What am I missing?

Schalonsus, 2017-02-22:
Can i compile my own F1 firmware with the time fix of F3 and just need to drop enough other features? There are a lot features in F1 i never use.

csurf, 2017-02-22:
@Schalonsus,
I took a look at some of the timer code, and it seems that they use the same math. The only difference is that the F3/F4 code uses a 64-bit integer, while the F1 uses 32-bit. Other than that, everything works the same. Someone please correct me if I'm wrong about this.

BTW, has anyone ever even tried running an F1 FC over 70 minutes to see what happens? Does the board lock up or something, or does the PID/task math just get screwed up while the roll-over happens and cleans itself up?

csurf, 2017-02-22:
As far as F1 features go, there's a bunch of crap that a lot of people don't even use, especially the fixed-wing guys (LED's & sonar come to mind). 

can we get a list together of the most ROM-consuming features? Perhaps we can get a vote going on which features should stay & go on F1.

martinbudden, 2017-02-22:
@csurf you say: 
> still not clear why this is even an issue. counter roll-over is a common issue that's been dealt with even since 8-bit multiwii days.

Yes, doing the roll-over is not complicated. What is complicated is testing all the edge cases so that we can be confident it works.

> BTW, has anyone ever even tried running an F1 FC over 70 minutes to see what happens?

There have been some reports of people flying over 70 minutes and not having problems. The issue is that we cannot **guarantee** that there are no problems.

I stand behind the code I write. I'm not prepared to stand behind 70 minute flight times on F1 targets based on wrap-around unless it has been well-tested. The effort of testing this is not small and I'd rather spend my time on implementing new features, especially since there is a solution for people who want longer flight times - which was put in at my instigation, by the way.

As far as F1 features go, LEDs and SONAR were dropped long ago.

But if you wish to organise a vote about what features should stay and go in F1, then that would be useful. I have no problem changing the feature set based on popular demand, and I'm pretty sure @digitalentity would be in agreement.

Redshifft, 2017-02-22:
@csurf  many features are pulled already (BB and LTM  remain) but I'm sure with a little imagination the problem can be addressed

digitalentity, 2017-02-22:
Some parts of the code still use absolute times in microseconds - RX code for sure, probably LEDSTRIP, maybe some other parts of the code as well.

digitalentity, 2017-02-22:
There are reports of people flying for over the 70min limit, but I can't guarantee that i.e. failsafe will work correctly over that limit.

Schalonsus, 2017-02-22:
Yes i did with 1.2 i believe.
But now since the discussion came up with flight time issues i'm abit afraid flying this long again.
Would it be possible to write a simulator for the code to test such scenarios?
Or maybe we can use x-plane in some way?
[image]

csurf, 2017-02-22:
@Schalonsus, 
even though others might not agree, I'm betting this can be solved & worked through without much issue. It'll require some code fixes and testing, but it's probably very do-able. 

If someone is willing, it might be worthwhile to work on a pull request that goes through all of the old code and fixes any instances of absolute time usage so that it can deal with micros() roll-over.

Anyway, I think maybe this should probably be opened up in a separate issue ticket. 

As far as F1 features go, can someone help with providing a list of features that are currently enabled for F1 builds in order of resource usage?

martinbudden, 2017-02-22:
@Schalonsus , it's fairly straightforward to do your own build, enabling the features you want.

Schalonsus, 2017-02-22:
@csurf ,
i'm not very good at coding, just a bit of java style. So i can not help with that.
@martinbudden ,
yes i know. I compile myself all the time because BMP085 is removed from naze target...
But is the 64bit code thing a feature i can enable? If so please point me to it.

martinbudden, 2017-02-22:
Just `#define USE_64BIT_TIME` in `common.h` under the `#if (FLASH_SIZE > 64)` line. You'll need to disable other feature(s) to provide enough ROM space.

Schalonsus, 2017-02-22:
This is awesome! Thank you alot for this information.

Redshifft, 2017-02-22:
@martinbudden  so how much extra memory will that little timing mod need ?

Redshifft, 2017-02-22:
@Schalonsus  impressive flight :) was that powered full time ?

Schalonsus, 2017-02-22:
That was way too easy 😋 
Just removed 
drivers/barometer_ms5611.c \
drivers/barometer_bmp280.c \
#define USE_BARO_MS5611
#define USE_BARO_BMP280

and added
drivers/barometer_bmp085.c \
#define USE_BARO_BMP085
#define USE_64BIT_TIME

fits easy in Naze target if you remove unnecessary stuff. I could have removed so much more. Thanks again for the hint @martinbudden 

@Redshifft ,
yes was powered all the flight time. Use Li-Ion cells...

martinbudden, 2017-02-22:
@Redshifft , when `USE_64BIT_TIME` is set, Naze target overflows by about 1200 bytes.

stronnag, 2017-02-22:
370 bytes here (gcc 6.3), iirc

csurf, 2017-02-22:
so, again, what features/codesets are taking up the most space on F1 builds? How much can be saved by removing unnecessary sensors and RX protocols?

stronnag, 2017-02-22:
We have already stripped F1 down as far as it can generally go without 'personnal' customisation (e.g. no RX other than the one _you_ fly). There is no general solution any more, we exhausted those options some time ago.

martinbudden, 2017-02-23:
@csurf , it is fairly straightforward to do your own build. You can work through the options in `common.h` turning them off one by one and doing a build each time to get this information, and so produce the list of information yourself.

Redshifft, 2017-02-23:
The custom build would be the best answer, I suppose the Afromini and Rev 5 would be the most popular "Naze" targets out there and worth their own hex ? there are files for much less popular boards built at every release.
Worth a poll as I think someone suggested :)

digitalentity, 2017-03-15:
I think F1s are still alive and this issue can be moved to 1.8

digitalentity, 2017-06-28:
NAZE is at it's limit. I think next release will be the last for NAZE target. Pity, but I guess it's time to say good buy to it. CC3D will stay for a little longer though.

stronnag, 2017-06-28:
https://github.com/iNavFlight/inav/wiki/Building-custom-firmware

may help (or just prolong the pain).

digitalentity, 2017-06-28:
Yes, those who wish to fiddle with removing features could still build custom firmwares.

giacomo892, 2017-06-28:
As soon CC3D is officially supported i'll still build my own build and they will work 100% as before :) 

Later with 1.8 I will continue to build but more testing will be needed. 

Hoping in a F1 community to test custom builds :)

BTW. Yes. I have bought some F3 boards too 💃

digitalentity, 2017-06-28:
I'm still using CC3D on some of my builds. IMO it's still one of the most reliable boards, although due to memory issues quite limited on functionality.

oleost, 2017-06-28:
@stronnag You may want to link to this page in that wiki. https://github.com/iNavFlight/inav/wiki/Making-a-new-Virtualbox-to-make-your-own-INAV

stronnag, 2017-06-28:
@oleost : done, thanks.

digitalentity, 2017-09-24:
F1's are no longer "officially" supported. Closing

closed 2017-09-24 by digitalentity