iNavFlight/inav #1297: Rudder in horizon/angle mode does not behave correctly

default-pt, 2017-02-16:
Hi,
I have a multiplex solius and added a F1 rev6 board with inav 1.6 and my issue is that on horizon mode the airplane behaves strangely.  If I turn to the right using ailerons, then the rudder counteract and turns to the left (looking from behind the plane). So, before stating this is an issue can I need to ask a question.
In horizon/angle mode if I bounce the plane to the right (nose to the right and tail to the left) the rudder should compensate to the left (viewing from behind)? or to the right?

Thanks .

oleost, 2017-02-16:
Hi.

> If I turn to the right using ailerons, then the rudder counteract and turns to the left (looking from behind the plane). 

I cannot reproduce this at all, are you sure that your radio is doing no mixing at all, and are you sure that rudder and elevator servo is correctly wired up ? ( Because it does mixing to the Elevator nav_fw_roll2pitch )

> In horizon/angle mode if I bounce the plane to the right (nose to the right and tail to the left) the rudder should compensate to the left (viewing from behind)? or to the right?

Yes

default-pt, 2017-02-16:
Hi and thank you for the reply.

So this brings me to the issue "If I turn to the right using ailerons, then the rudder counteract and turns to the left (looking from behind the plane)." 
Completing the sentence, when flying if I bank right using the ailerons, naturelly the plane tail will move to the left and starts to turn right.. the thing is that then the rudder will counteract to the left pushing the tail to the right and will not let the plane do a good turn. 

Regards

oleost, 2017-02-16:
Hmm, this was an issue that was supposed to be fixed a while ago. Guess we will have to wait for an answer from @digitalentity 

Could you provide both CLI dump and ideally also a blackbox log of an flight with aileron turn as you described.

default-pt, 2017-02-16:
Thanks, while we wait I will try to get the blackbox this weekend.

digitalentity, 2017-02-17:
@default-pt @oleost this was partly solved for navigation-enabled modes - they do rudder/aileron coordinated turns correctly.

In Angle/Horizon mode bank and yaw are decoupled, so plane does exactly what it is commanded to - with aileron stick you command it to bank right, but you don't command a right turn using yaw stick - so the plane does its best to prevent the "uncommanded" turn.

This is how it was behaving from the very beginning. A temporary solution is to set Yaw I = 0, or get used to do banked turns with two sticks as on quads.

We should probably look into extending TURN ASSIST logic on airplanes to do proper rudder/aileron coordinated turns towards 1.7

oleost, 2017-02-17:
Ah ofc. Also makes sense to keep it as is, for acrobatic airplanes etc.

Would need to be implemented in example as `TURN ASSIST` as you mentioned.

default-pt, 2017-02-17:
Hi.
Thanks for the explanation. So in navigation modes like poshold and  althold  if I turn with aileron to the right then it will do a coordinated turn and if I release the sticks it will automatically level?

Thanks

default-pt, 2017-02-17:
Also, is there any way to disable the rudder on horizon mode but not on passthrough?

default-pt, 2017-02-17:
Hi,

So my resolve is to reduce P yaw from 85 to 10 and increase the yaw rate from 200 to 700.
With this change the counteract from the rudder is minimized and with the increase of the yaw rate I get back the rudder authority that I lost from the reduction of the P yaw. Hope this works on the field.

The answer for my question to disable the rudder on horizon mode (but it will affect other modes also) is just to cut completely the P yaw to zero and as passthrought mode states when selected it will let all servo signal pass so we will still get full rudder control.

And all navigation modes begin with NAV in the name... silly of me.

But the issue stated before : 
"when I fly the plane with horizon mode and do a coordinated turn using the ailerons and rudder, the rudder still moves a bit (but less than if i do not use the rudder) and counteracts the turn... even if we think in aerobatic flight if we use the rudder stick then there should not be any help from the FC, only when we let go the sticks and they are center like for example between 1498 to 1501."

still makes sense to correct...

Regards

digitalentity, 2017-04-05:
Partly fixed by #1512. Will be fully fixed when full airspeed support is introduced

closed 2017-04-05 by digitalentity