iNavFlight/inav #1299: blackbox_rate _denom > 32 does not behave as expected.

stronnag, 2017-02-16:
As discussed on RCG:

With `blackbox_rate_denom = 50` (50, 14 etc)
````
loopIteration	 time (s)
0	35.907236
50	35.957814
64	35.972139
114	36.022701
128	36.036887
178	36.0875
192	36.101595
242	36.152211
256	36.166564
````
With `blackbox_rate_denom = 192` (192, 64  etc)
````
loopIteration	 time (s)
0	21.404023
192	21.598367
256	21.663182
448	21.857686
512	21.922587
704	22.117116
768	22.181896
960	22.37625
1024	22.440921
````

digitalentity, 2017-02-17:
Blackbox logs several types of frames - flight behavior is written using I- and P-frames. I-frames are fairly big and contain absolute values, P-frames are delta-encoded to save space. Blackbox denominator only reduces P-frame rate, I-frame rate is constant.

Originally I-frames were logged every 32 iterations, P-frame is logged every `blackbox_rate_denom` after I-frame. This doesn't give you exactly `1 / blackbox_rate_denom` rate, i.e. for 1/16 - 1/31 rates it's going to be c. 16 iterations between frames on avereage.

New code choses I-frame interval dynamically of 1/32, 1/64, 1/128 and 1/256 based on `blackbox_rate_denom` used.

1/50 rate you used choses 64 as I-frame interval, meaning c. 1/32 actual logging rate.

Possibly we need to change the logic for rates slower than 1/32.

stronnag, 2017-02-17:
I've now added the explanation above to the wiki, https://github.com/iNavFlight/inav/wiki/INAV-blackbox-variables ; I think this is good enough and we can close the PR.

closed 2017-02-17 by stronnag

digitalentity, 2017-02-17:
@stronnag thanks