iNavFlight/inav #1300: Autopilot enhancement: nav_fw_bank_rate paramter to avoid jerky AP behavior

Dronek, 2017-02-17:
As explained [here](https://www.rcgroups.com/forums/showpost.php?p=36894819&postcount=11335) it would be nice to limit the rate at which nav_fw_bank_angle is reached in order to maintain smooth aircraft behavior in nav mode (i.e. turning to next WP or 180 deg turns after triggering RTH). Similar rate parameters are imaginable for dive and climb too but bank_rate would certainly be the most useful one for fixed wing.

Please close with a short comment when this functionality can be achieved by other parameters.

oleost, 2017-02-17:
https://youtu.be/50C61gqCBSQ?t=5m15s

oleost, 2017-02-17:
Maybe try and reduce self leveling p-gain `p_level`

Dronek, 2017-02-17:
Will p_level affect steering input from AP?

oleost, 2017-02-17:
Ap?

If you mean GPS modes like WP etc then yes.

This adjusts self leveling in angle mode. And all other modes than acro is based on angle mode ( Just like angle works on top of acro mode ).

But I gain should take the rest.

digitalentity, 2017-02-17:
@Dronek @oleost 
`LEVEL P` controlls the speed of response to angle command. The higher it is - the faster the machine will try to reach the commanded angle
`LEVEL I` is the low-pass frequency of angle response. If you want it smoother (but still fast) - reduce this down to 1 (zero disables)

Dronek, 2017-02-18:
Thank you both. Will try and report :)

Dronek, 2017-02-24:
>  LEVEL P  controlls the speed of response to angle command. The higher it is - the faster the machine will try to reach the commanded angle
 LEVEL I  is the low-pass frequency of angle response. If you want it smoother (but still fast) - reduce this down to 1 (zero disables)

Still no chance to fly due to stormy weather. Will close this issue and re-open if the method above doesn't help.

closed 2017-02-24 by Dronek