iNavFlight/inav #1301: Which navigation modes work for Airplane and Flying Wing on v1.5

default-pt, 2017-02-17:
Hi,

I'm having a difficult time to identify which flight modes are working/meant to Airplane and Flying wings. There is no clear statements that can help to check what modes work or not.

From the existing ones (if there is more please add) can anyone help identify:

AIR MODE - ???
ANGLE - makes sense
ARM - makes sense
BEEPER - makes sense
BLACKBOX - makes sense
CAMSTAB - ???
FAILSAFE - makes sense
FLAPERON - makes sense
HEADADJ - does not make sensse
HEADFREE - does not make sense
HEADING LOCK - Only if MAG mode makes sense
HORIZON - makes sense
LEDLOW - ???
MAG - should make sense, but the information that exists states that only GPS and Baro is needed
NAV LAUNCH - Check
OSD SW - Check
PASSTHRU - Check
SERVO AUTOTRIM - Check
SURFACE - ???
TELEMETRY - Check
TURN ASSIST - ???

Also the configurator should only show modes that can be used for the type of aircraft that is selected.

Can you please check?

Note: I can help updating iNav documentation regarding this to make clear to others.

Regards

flounderscore, 2017-02-17:
For AIRMODE, I'd even say it's required for fixed wing to keep the control
loop running at 0 throttle.

default-pt, 2017-02-17:
So it should be added to all flight modes?

sppnk, 2017-02-20:
I said this before, but IMHO, AIRMODE is a weird thing for FW. It should be removed or hardcoded for fw.

DzikuVx, 2017-05-23:
Firmware allows Configurator only modes that can be enabled on a Fixed Wing or Multirotor.

closed 2017-05-23 by DzikuVx