iNavFlight/inav #1303: Re-license INAV under less-restrictive open-source license

digitalentity, 2017-02-18:
GPL is a very restrictive license. It essentially prohibits a few things that might be very benificial to the project:

1. Linking against libraries available in binary form (closed sourced or with source available only under NDA)
1. Including non-GPL code (Code, which is open, but not licnesed under GPL). If author of this code is not willing to dual-license it under GPL - this source code is prohibited from being included in INAV
1. Possibility of licensing some INAV technology to be used in commertial applications (currently will allow 100% of the resulting product to be opensource under GPL which is essentially a show stopper).

All these issues can be solved by re-licensing the source under a more permissive open-source license, i.e. Mozilla Public License (MPL), which puts very light restrictions about combining open-source and closed-source code.

The MPL treats the source code file as the boundary between MPL-licensed and proprietary parts, meaning that all or none of the code in a given source file falls under the MPL. Licensed source code can be combined with other files under a different, even proprietary license, thereby forming a "Larger Work" which can be distributed under any terms, but again the MPL-covered source files must be made freely available.

This way we'll still have the protection against someone taking the source and claming it as his property, while having a possibility of using differently licensed libraries in our project.

### I want to emphasize that INAV will stay open-source and free.

From `git` commits I've figured out that there are just 67 contributors in the INAV source, just 17 of which "own" over 100 lines of code.

I'm asking these 17 major contributors - do you allow re-licensing your work (code) within INAV under terms of Mozilla Public License?

List of contributors:

- [ ] @digitalentity
- [ ] @martinbudden 
- [ ] @hydra
- [ ] @sambas 
- [ ] @jflyper
- [ ] @thenickdude
- [ ] @ledvinap
- [ ] @DzikuVx 
- [ ] @gaelj 
- [ ] @4712
- [ ] @frank26080115
- [ ] @ProDrone
- [ ] @marbalon
- [ ] @nzmichaelh
- [ ] @MJ666
- [ ] Stefan Grufman <stefan.grufman@gmail.com>
- [ ] @Pierre-A
- [ ] @stronnag 
- [ ] @ethomas997
- [ ] @andre92
- [ ] @Linjieqiang
- [ ] @borisbstyle
- [ ] @oleost 
- [ ] @hali9
- [ ] @Faduf (probably an alias of @francoisduf)
- [ ] @DarkVegetableMatter
- [ ] @airmamaf
- [ ] Richard Lehey <rlehey+gh@gmail.com>
- [ ] @ChiggerPepi
- [ ] @sneiman
- [ ] @SteveAmor
- [ ] @flacjacket
- [ ] @samcook
- [ ] @fedorcomander
- [ ] @avoid3d
- [ ] @kilrah
- [ ] @lkaino
- [ ] @tracernz
- [ ] @theArchLadder
- [ ] @larryho5
- [ ] @krzysztofmatula 
- [ ] @St7ven
- [ ] @projectkk2glider
- [ ] @nebbian 
- [ ] Carsten Wache <lichtl@x-modder.de>
- [ ] @goebish
- [ ] @rimasaviz 
- [ ] @basdelfos
- [ ] pinkywafer@users.noreply.github.com
- [ ] @chris-penny
- [ ] @MotoLab
- [ ] @blckmn 
- [ ] @readerror67
- [ ] @MauroMombelli
- [ ] @rosek86
- [ ] @fhorinek 
- [ ] @Steveis <steveis@stevis.com>
- [ ] @Netsplits
- [ ] Florian Huber <flo.g.huber@gmail.com>
- [ ] @fishpepper
- [ ] @MasterZap
- [ ] @DaTwo
- [ ] @sppnk
- [ ] @Philippe-France

digitalentity, 2017-02-18:
Mentioned people, please kindly leave a comment here with your authorization to re-license your contribution to INAV under MPL. You will still be considered as author of the code and legal copyright holder of your contrinution, this is NOT taking away your rightfull ownership.

stronnag, 2017-02-18:
I do NOT consent to my (limited) contribution being relicenced, nor do I agree that cherry picking this list of contributors from the prior works which would include baseflight and multiwii is either legal or ethical.

digitalentity, 2017-02-18:
@stronnag may I ask why? Re. list of contributors - this list is built from complete `git blame` and reflects only authors who's contribution is still present in the source tree.

stronnag, 2017-02-18:
Because I'm opposed to the inclusion of non-free, closed source code in open source projects. I'm also surprised that  all prior code has been replaced and I'd be interested to know how that audit has been performed. The licence does not just apply to code post the fork from cleanflight.

digitalentity, 2017-02-18:
I understand your point. So this is probably a no-go for the re-licensing.
The audit was performed by parsing output of `git blame` excluding comments, `#include`s and empty lines:

stronnag, 2017-02-18:
But that superficial analysis excludes original copyright holders. Consider these two small contributions:

* LTM: The real copyright holder is KipK, the Inav contribution just refactored it for CF and iNav
* Spektrum RSSI: The real copyright holder is the author of the original BF patch,  the Inav contribution just pasted it into iNav.

I doubt these are isolated cases.

digitalentity, 2017-02-18:
Good point, it really skipped my mind.

goebish, 2017-02-18:
Same boat here, for example when I added the v202 protocol for nrf24l01 RX, I copied the code from Victzh's port of bradwii.

digitalentity, 2017-02-18:
Bottomline - re-licensing is probably not possible. Shame. This makes effort of integrating into industrial-grade systems more complicated. We'll have to find another way to co-exist with proprietary modules in one system.

stronnag, 2017-02-18:
Or, depending on your point of view, excellent the GPL again protects free software. We can agree to disagree on this one :)

digitalentity, 2017-02-18:
I totally agree that GPL is the best of FLOSS licenses to protect freedom. But it does so at expense of significantly limiting possibilities to evolve the software as a product.

There is a talk about evolving INAV into an industrial-grade autopilot - GPL'ed code makes this significantly less possible. This is very unfortunate (of fortunate, we'll wait and see). :smile:

borisbstyle, 2017-02-18:
I agree with @digitalentity and i totally support his idea. 
@stronnag unfortunatelly the current code is being used a lot in other closed projects and its hard to see some of your hard work being abused on that way.

digitalentity, 2017-02-18:
@borisbstyle, I was just writing a possible scenario where GPL will bite us instead of mis-users.

Let's say some Chinese company takes INAV, puts it into own autopilot and sells it as a "CoolFlyPro v3.14" without even stating that they used an open-source code there. They benefit from what we do and we don't even get the fame. GPL doesn't protect us if we don't know that our rights are abused.

nebbian, 2017-02-18:
For the record, I have no problem with anyone using my code, for whatever purpose they feel like, or relicensing it under whatever scheme they choose.  Go for it.  Knock yourself out.  Make great products.  Make the world a better place.   I'm just happy that I have helped the community in some small way.

DzikuVx, 2017-02-18:
Personally I have no objections against relicensing. No possible? Well...

nzmichaelh, 2017-02-18:
For reference, I'm happy for my contributions to be relicensed to MIT, MPL, or similar.

flacjacket, 2017-02-18:
I am fine with relicensing to a weak copyleft license

MJ666, 2017-02-18:
I have also ported some code other sources (Harakiri, BaseFlight, MultiWii etc.) in the past and can not realy talk for this parts. There I likely also quite some code in INAV where the orginal autor does not appear in the list of contributors from the time before GIT was used like all the MultiWii autors. Likely relicensing is not an easy thing to do. For my personal contributions I would agree for relicensing.

fedorcomander, 2017-02-18:
For tiny bit of work i made for CF - i am fine to change license.

ChiggerPepi, 2017-02-18:
Only an itty bitty contribution by me, but I'm okay with whatever happens with the liscensing.

neonbjb, 2017-02-18:
Same here - my contribution is small and I don't care if the license is
changed.

thenickdude, 2017-02-18:
I do not consent to the license of my code being changed to MPL. If commercial organisations would like to benefit from my technology without giving us the benefit of theirs, that's fine, but they will need to reimplement it from scratch without reference to the original code, by reading this design document:

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Blackbox%20Internals.md

I'm happy to clarify anything that the document leaves ambiguous.

fedorcomander, 2017-02-18:
Nicholas: Its MORE then enough to implement server side of things... Thanks mate!

digitalentity, 2017-02-18:
@thenickdude those docs on BB are very thorough - should be more than enough to implement BB encoder using them!

Linjieqiang, 2017-02-19:
Forgive me for being so late just because I went to a trip in the past days.I support your idea about "Re-license INAV under less-restrictive open-source license".Also,I will pay attention to this project and will be willing to make a contribution.Thanks your great work again.

Faduf, 2017-02-19:
I'm fine to relicensing.

Faduf

projectkk2glider, 2017-02-19:
Fine with me.

oleost, 2017-02-19:
Fine for me.

chris-penny, 2017-02-19:
ok for me too, my input was minimal and I mostly don't understand the intricacies of the possible issues between the different licences.

basdelfos, 2017-02-19:
I have no problem with changing the licence.

tracernz, 2017-02-19:
In general I would prefer my C/C++ source code to remain GPL so I, and others, may enjoy free use of improvements built on top of it. In this case my contribution is relatively small so I have no intention to block; you may re-license the small portions I contributed.

It might help others if you were to explain your true intentions rather than a a few vague suggestions of possibilities? Also, the problems raised earlier regarding 3rd party and misattributed code would need to be addressed, especially code contributed under MultiWii.

digitalentity, 2017-02-19:
People seem to think changing from GPL to MPL is a path to completely closing the source - it's not. If we leave the most significant parts in both licenses we'll see that both require the source to remain open and free (as well as the changes made to it). The difference is the boundary between free and non-free code. GPL requires the whole software thing to stay free and open-source. MPL limits this requirement only to specific source files remaining free and open-source.

If we change from GPL to MPL the whole current source tree will remain free and open. We only weive the limitation to include closed-sourced (or differently licensed) parts here.

digitalentity, 2017-02-19:
@tracernz you wanted to know my true intentions - they are to keep INAV evolving to better, more robust AP, suitable for industrial use (eventually). We already have some proprietary code we might benefit from (https://github.com/iNavFlight/inav/issues/962#issuecomment-276969039) at least temporary, until someone re-implements the algorithm. 

It's not the only piece of code that will never make it into INAV because of GPL. I have a few aces up my sleve - compass-less operation on multirotors being one of these. However, due to unfair competition from Chinese manufacturers who don't care about licenses and patents I'll probably never release this code under any open-source licnese (at least not at the moment).

@stronnag was also very clear about inclusion of closed-source code into INAV, so these possible low effort improvements (I would even say unique features) will have to stay away from INAV.

digitalentity, 2017-02-19:
We also still have the problem about misattributed code originating from MultiWii which we probably won't be able to address - it's virtually impossible to track those parts of code.

sppnk, 2017-02-20:
Changing the licensing is ok for me too, (altought my contributions to the code have been tiny :-)).

andre92, 2017-02-20:
Please remember this is the legacy of MultiWii. In my opinion, if you want to relicense, you need to rebuild iNav from scratch, just like Raceflight did. I don't care about my contributions, do whatever you like, but I am against relicensing a project that dates back over 5 years.

KipK, 2017-02-20:
Sup.

Considering I've been out of the UAV community for a while, and dunno If I'll have time to get back in one day, feel free to relicense my code if it can help.

Best.

digitalentity, 2017-02-20:
Several key developers have got back to me with explicit "NO" for re-licencing, including some of original developers of MultiWii code.

I'm closing this issue now. We'll have a different way of doing more cool things - it will not be pretty, but it will be GPL-compliant.

digitalentity, 2017-02-20:
Huge thanks for those who allowed me to re-license their contribution. I really appreciate your openness!

hydra, 2017-02-20:
Bit late to the party, but my answer is No.  I wouldnt feel happy about it and certainly not without also knowing the the thoughts from the people's work that was used to make it what it is.  Without code being open people wouldnt be able to create things as easily using ideas from other open projects. This project borrows heavily from many sources.

digitalentity, 2017-02-20:
@hydra you are not late, talk is still open. Do you really think MPL will allow closing the code? You said we are borrowing from many sources - are those sources available under GPL-compatible licenses? Aren't we violating those licenses ourselves?

digitalentity, 2017-02-20:
Regarding the original MultiWii code - determination of copyright is usually based on new work's similarity to original work. We are at the point where we possibly don't contain a single line of code from original MultiWii. 

Now, do original MultiWii authors still hold copyright to the code that was at some point back in time based on their code but since then was literally re-written and doesn't contain their work any more?

andre92, 2017-02-20:
@digitalentity when I wrote my answer I was thinking about code that has been rewritten but still does the same, for example the stick sequences for arm/disarm, profile switching, triggering calibrations and eeprom write etc, or the multiwii serial protocol, motor layout. I am not sure at which point we would be free of MultiWii legacy.

borisbstyle, 2017-02-20:
@andre92 Yes correct ........the only few ugly parts left of the code are still multiwii. Just the stick command code and the painfull MSP :D

fhorinek, 2017-02-20:
I "own" only 2 lines of code that anyone could write, so do what you want.
I do not care about the licence, but if it means that I won't be able to fiddle and recompile the sources and get fully working firmware, my answer would be No.

digitalentity, 2017-02-20:
This is complicated. If I take the specs of MSP and written my own implementation will this be considered a derivative work? Probably not. We have done precisely that - our MSP implementation is completely different under the hood. 

Ideas can't be copyrighted, so if our code is different it's free from legacy even if it's doing the same thing - that's my understanding and I might be wrong.

Probably the only thing that can still be traced back to MultiWii - that's stick combo code.

borisbstyle, 2017-02-20:
@digitalentity You are right. The current MSP code has been so refactored and changed that it doesnt even come close to original MSP.

But the stick combo code is like exactly the same :D

digitalentity, 2017-02-20:
@fhorinek thanks. MPL is also an open-source license that doesn't allow closing the source. Everithing that is open at the moment will stay open.

digitalentity, 2017-02-20:
@borisbstyle hence my original question - should we consider MultiWii at all? Should we just re-write the stick combo processing part and be free from 7-year legacy?

andre92, 2017-02-20:
@digitalentity that would work. When we are free of legacy I wouldn't mind changing the license.

borisbstyle, 2017-02-20:
@digitalentity It happens that the rx code part was next on my list to do anyway :D. The higher level rx code has all fully changed to floats. My next step was to simplify the rcdata processing together with the leftovers of multiwii stick combos and few other rcCommand pre processing parts.

fhorinek, 2017-02-20:
@digitalentity But does it allow it to bundle the open code with closed one in a way that you can't compile the open code without the closed code?

digitalentity, 2017-02-20:
@andre92 thanks. People seem to misunderstand my intentions and think that I'm trying to make INAV closed source or render it useless without proprietary parts. Guys, this is only going to happen **over my dead body**!

If we manage to pull this off - some parts of INAV (for very specific features like mag-less operation, RTH by own tracks etc) certainly will be proprietary, but this will never make INAV unflyable without them and I'll do my best to make them available to all who uses INAV in a legitimate way.

MauroMombelli, 2017-02-20:
AFAIK the example of defending against the company coping the code is noto possibile with any licence.

Please note GPL code accept the use of proprietary code IF it is a system library (so if user want to use its own hardware and does not want to open source that implementation should be fine.. Even if it is very hard now to draw a line) or if it is an external plugin. So to avoid relicense you could create an api and let the proprietary act as a plugin.

As side effect any company that bugfix the core application or improve the api MUST report back any modification. This will also insure the core will remain flyable.

Oh, I don't have any code publish, I partecipeted and worked on the DCM implementation on cleanflight

digitalentity, 2017-02-20:
@MauroMombelli - unfortunately you can't make a plugin and link it with GPL software and get away with it.

I'll quote FSF here:
>If the program dynamically links plug-ins, and they make function calls to each other and share data structures, we believe they form a single program, which must be treated as an extension of both the main program and the plug-ins.

Simply speaking this means that sharing the data in common memory makes two parts of the code one big program and it should be available under GPL.

GPL is very restrictive license - all parts of the software must be free and opensource and under GPL.

4712, 2017-02-20:
I'm fine with re-releasing my small part of work (4way-if) under MPL or dual license or whatever. 
But I cannot decide for others who cleaned up my code... (@ledvinap, @martinbudden, @nathantsoi ...)

krzysztofmatula, 2017-02-20:
My contribution is not big so far, and I agree to change the license into a less restrictive version of open source. It's good it will open new possibilities. It's good if it will make the project more available to the industry. The more users (and testers) the better the code. In fact, there are "workarounds" to use GPL code in commercial products anyway.

gaelj, 2017-02-20:
No problem to change the license for me.

hydra, 2017-02-20:
@digitalentity For me it's all about keeping it open, and by 'it' i mean everything we've done so far **and will do in the future**

This is likely the intent of others who chose the original license and I don't want to go against that decision, I like what we've created and want to see it continue, in the open.

Alexinparis, 2017-02-20:
Hi,

Some steps back in history:
I wrote the first versions of MultiWii from scratch and alone.
It was called TriWiiCopter before supporting other multiple motors configs.
I took some inspiration (not code) from 2 other projects: Shrediquette and
AeroQuad.
Note this explains also the historical choice of 8bit AVR.
For sure, I wouldn’t have been able to develop so fast at this stage
without the possibility to read real code that works (many projects, many
blogs, some forums, …).
When I decided to open the resulting code, the GPL choice was quite obvious.

My main motivations were:

-          share code and algorithms knowledge

-          no possibility to enhanced the code without sharing it

-          a sort of entropic growing functionalities from other people
sharing the same view

-          no restriction on the way to use it : probably the most
important point for me. Including any commercial usage. At this time I
hoped a drastic and uncontrolled reduction of FC board price and it
happened :)

Now I don’t own any more MultiWii code, and I can give only my opinion on
this topic.

Due to the number of further contributions based on the original code, I
think it’s now impossible to change the license.

1-  Linking against libraries available in binary form (closed sourced or
with source available only under NDA)

From linux world, I know it’s possible to write and run a close source
driver as a binary inclusion. So I suppose it’s possible with any GPL
source project. Probably not as simple as “and they make function calls to
each other and share data structures”

2- Including non-GPL code (Code, which is open, but not licnesed under
GPL). If author of this code is not willing to dual-license it under GPL -
this source code is prohibited from being included in INAV

It’s the principle of the license…

3- Possibility of licensing some INAV technology to be used in commertial
applications (currently will allow 100% of the resulting product to be
opensource under GPL which is essentially a show stopper).

Many companies are doing business on GPL code as a tool (service, hardware,
use case,…). I approve it.
But one on my motivation was to reduce possibilities to do business on the
code itself, as an IP.
It can be probably bypassed in some way with 1- strategy or with a deep
code refactoring avoiding any similarities. It’s not really my view and I
can’t go against.

[21 more replies]

closed 2017-04-09 by digitalentity