iNavFlight/inav #1306: in-flight 'board orientation' adjustment for trimming level flight

csurf, 2017-02-18:
Does this already exist? 
A way to perform an in-flight trim of the aircraft for level flight, for flight modes that require self-leveling (ANGLE, HORIZON, etc.).

Back in CF/BF, you could do this via the acc trims. However, iNav relies on the board orientation as a reference for level flight. As far as I know, there isn't a way to fine-tune the board orientation in flight (or is there?)

If this isn't available, can we please add this as a feature request? I think the best/easiest way would be to implement it in the exact same way that 'servo auto-trim' works for fixed-wing. thoughts?

csurf, 2017-02-21:
bump on my question...
@digitalentity, would something like this be possible to setup?

Dronek, 2017-02-21:
It seems that the feature, you're asking for, was intentionally [removed](https://github.com/iNavFlight/inav-configurator/issues/125).

As far as I understand "servo auto-trim" is used to trim the servos for straight flight in acro mode so that PID/FF output is minimal without external disturbance or pilot's input.

csurf, 2017-02-22:
yes, I understand servo auto-trim, but it doesn't serve the same purpose as having to adjust the aircraft for level flight in HORIZON/ANGLE mode (and I believe also for the NAV modes?). 

The base/clean/betaflight way of doing it was via acc trim setting, but iNav relies on board orientation. It's far too clunky having to try to figure out the correct pitch&roll angles for level flight, land, connect to configurator, adjust the board orientation, fly again, test, repeat, etc. This is far too complicated and takes too long. 

It would be far easier and better to be able to fine-tune the board orientation angles for pitch&roll in-flight in a similar manner as how servo auto-trim works: 
- launch aircraft
- go into ANGLE or HORIZON mode
- move sticks so that aircraft flies level
- toggle on a preconfigured 'LEVEL TRIM' mode switch
- average pitch&roll angle values after 'x' seconds are incorporated into board orientation 
- if you want to try again, toggle the 'LEVEL TRIM' switch off, set sticks, toggle on, rinse & repeat, etc.

Does this seem reasonable?

digitalentity, 2017-02-22:
The thing is that board alignment is applied to raw sensor data (prior to filtering). Changing that in flight may cause very unexpected behavior.

csurf, 2017-02-22:
@digitalentity 
ok, understood.
can you suggest some other way to possibly achieve the same goal? perhaps a temporary 'trim' value that can be stored until the board is disarmed, and then it's applied to raw board orientation? just brainstorming here...

Redshifft, 2017-02-22:
- Recalibrate the board at Aeroplane (flying level) angle
- Adjust linkages and balance of Aeroplane to trim, redefining servo center is a bad thing anyway as they should have been set up mechanically in the first place.

If you don't have an hour to spend optimising trim, you have the wrong hobby :)

csurf, 2017-02-22:
cute, but that's not very helpful.

The point is to aim for efficiency and improve on existing methods, and not regress on key features. Otherwise, what's the point of all of this development anyway? Almost all FC firmwares that I know of offer this feature is some form, so I don't think it's an unreasonable concept.

cdog4twenty, 2017-09-09:
This is a very valid question, coming from a vector. It has in flight level calibration specifically for this. What if the fc isnt exactly flat or is turned the slightest. How can you compensate for this?

closed 2018-05-27 by stale[bot]

flyandride, 2018-12-21:
It is possible, watch this video.  The Irish accent is hard to understand, I have to replay sections to get it.  It's working on the bench so should work in the air, hopefully.  
https://www.youtube.com/watch?v=dQhO-SnpVoc

shellixyz, 2018-12-21:
@flyandride I didn't watch the video but a way to trim level in flight has already been added back in 2.0