iNavFlight/inav #1316: BMP085 is needed for AfroMini rev0

Schalonsus, 2017-02-20:
Please put BMP085 Baro back in Naze Target... it is needed for AfroMini rev0

digitalentity, 2017-02-21:
Doesn't fit. What should we sacrifice?
```
Linking NAZE
c:/cygwin/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld.exe: obj/main/inav_NAZE.elf section `.data' will not fit in region `FLASH'
c:/cygwin/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 374 bytes
```

Schalonsus, 2017-02-21:
I can't speak for all F1 users, but i don't need LED, nor Sonar...
Since these both are frequently requested, dropping them would be no good idea.
Afromini does not use MS5611, just BMP085, BMP180 and BMP280. Maybe Afromini gets its own target?
Compiling myself is no problem actually, but maybe it is a good idea to make an online compiler (if this is even possible?) so users could select their hardware and telemetry links and get it compiled.

handsomejackuk, 2017-03-03:
yeah would like also naze32 with bmp 180 or bmp 085 have tried version 1.6 and here that bmp180/ 085 is now gone.. i have no need for sonar or leds... if that could be removed... any tips on how to compile my own code please...

stronnag, 2017-03-03:
There is a whole section in the iNav docs (development) on how to build your own hex file, for various platforms, Linux, MacOS, Windows.

handsomejackuk, 2017-03-03:
thanks i have a look...

DzikuVx, 2017-08-04:
Since NAZE will not appear in INAV 1.8 I think we can close this one

closed 2017-08-04 by DzikuVx