iNavFlight/inav #1317: External compass calibration on iNav v1.6-RC1

raul-ortega, 2017-02-20:
I tried the compass calibration for this [GPS Ublox NEO M8N + Compass](http://www.banggood.com/Ublox-NEO-M8N-Flight-Controller-GPS-with-Protective-Shell-for-PIX-PX4-Pixhawk-p-1005394.html) (see link to product) without success on a ominibus f3 aio board.

The compass is detected as HMC5883, and after the calibration process the 3D model of iNav-configurator drifts towards left and right, and the angle value is erroneous.

The yaw axis of the board is alligned 270 degrees. The arrow of the GPS is pointing towards front, and mag alligment is default on configurator.

andre92, 2017-02-21:
I am guessing the mag sensor is aligned differently in the gps sensor. Sometimes they are aligned sideways, backward or upside down. You need to try all options for align_mag. With the correct option and the mag calibrated, the heading will remain constant when you pitch and roll the craft.

bascomptelab, 2017-02-21:
I live in the northern hemisphere. I tried all positions except CW0, I will try it this afternoon and will report again.
I recorded this vídeo, so you can see how it drifts: [https://www.youtube.com/watch?v=iUfuz5VT2lk](https://www.youtube.com/watch?v=iUfuz5VT2lk)

digitalentity, 2017-02-21:
At that GPS module the compass sensor is a) not straight and b) on the bottom part of the board. You have to calibrate the compass and then play with mag alignment until you get it working without drift. Probably CW90_FLIP or CW270_FLIP is going to work for you.

raul-ortega, 2017-02-21:
Now it works fine with CW0-flip (The omnibus is rotated around yaw 270 degrees).

Thanks so much.

closed 2017-02-22 by digitalentity

skaman82, 2017-02-22:
is there actually a way to check compass settings for correct orientation without trial and error?

digitalentity, 2017-02-23:
@skaman82 no, not really. We need external feedback how the copter is oriented in order to calculate the alignment automatically.

Kyle5G8, 2017-12-06:
Sorry to hit such an old thread... but I'm at a loss with my sp racing f3 and the se100 module from radiolink.

I'm having the same drifting problem, where as soon as I stop rotating the craft, it drifts back to its original heading in the setup tab. This same thing has happened with both the on-board magnetometer, and the external magnetometer with the on-board one disabled. I also notice that despite sturdy connections, the magnetometer in inav shows red when the craft is upside down. I've tried all alignments with no luck. 

The external mag responds to changes correctly but just doesn't hold its heading, but the inav sensor graph shows flat lines on all three axis. Any tips before I call it a bad board?

alanfj, 2018-02-07:
Your compass is a QMC5883L, not the Honeywell HMC5883, which is why it does not work. digitalentity is in the process of having it added to the next release coming soon. Here is a copy of the manufacturer document.
[QMC5883L-Datasheet-1.0.pdf](https://github.com/iNavFlight/inav/files/1705096/QMC5883L-Datasheet-1.0.pdf)   Gp to   https://nam03.safelinks.protection.outlook.com/?url=https://github.com/iNavFlight/inav/pull/2699&data=02|01||6a13dd69912242c7db5708d5661b83b8|84df9e7fe9f640afb435aaaaaaaaaaaa|1|0|636527190708243449&sdata=GerdkbKCgmuKtWdKaSrDyNxqlu2cByWNazsTMGh4lN4=&reserved=0