iNavFlight/inav #1318: LAUNCH mode with Auto Arm on iNav 1.6-RC1

raul-ortega, 2017-02-20:
I tried LAUNCH with fixed_wing_auto_arm without success in 4 attemps. MOTOR_STOP when arming is ON.

[https://www.youtube.com/watch?v=_Mrquiddfkg](https://www.youtube.com/watch?v=_Mrquiddfkg)

oleost, 2017-02-21:
Need more information.

Did you follow instructions on how to use launch mode (how did you do it)

What are your values from CLI dump?
Do you have an blackbox recording of those failed launches?

digitalentity, 2017-02-21:
Do you have a beeper on your wing? 

My two best guess: 1) misconfiguration (dump required to verify); or 2) misuse. Did you raise throttle after arming but prior to launch?

raul-ortega, 2017-02-22:
Unfortunately I forgot to insert the sd card, and I don't have installed the buzzer yet. I think that I followed the launching sequence, but maybe I did something wrong. Please could you take a look to my configuration parameters? Thanks so much.

[dump-1.6-rc1.txt](https://github.com/iNavFlight/inav/files/791874/dump-1.6-rc1.txt)

kposadowsky, 2017-02-22:
For using the launch assistent you should definitly have a buzzer on your wing - when you turn it on the right way it will sound like a ticking bomb.  Only then you can be sure the la is in charge.

You should at least set small_angle = 25 to 180.

digitalentity, 2017-02-22:
@kposadowsky you are certainly right. Without a buzzer you can't distinguish between disarmed plane and armed plane with motor_stop. In @raul-ortega's case the plane might have been not armed in the first place.

raul-ortega, 2017-02-22:
I agree with you about the need of a buffer, this is something that I'll solve before flying again.

The question is that I had the fixed_wing_auto_arm param set to on, and the numbers of sats where 11 on my goggles, so let suppose the plane to be armed.

Reading the at the wiki the sequence for launching the two first steps say:

1. Set switch to NAV LAUNCH mode prior to arming (note that it won't actually enable until arming)
2. ARM the plane. Motor should start spinning at min_throttle (if MOTOR_STOP is active, motor won't spin)

In my case with fixed_wing_auto_arm set to on It doesn't match the sequence because it arms before turn on the nav launch mode switch, then I'm inverting the sequence. Could this be the problem?

oleost, 2017-02-22:
It arms when you throttle up, so it should work.

1. Enable NAV launch.
2. Set throttle to 50%
3. Throw

One easy way to verify your ready is that start with passthrough mode, when you throttle up the servos should start to stabilize the wing when you move it.

sppnk, 2017-03-05:
Just to report that I am testing autolaunch in several planes (small and big ones) with great success. The MiniTalon is launched with rubberband and all the parameters are default, not touched. It works fine.

digitalentity, 2017-03-06:
@sppnk cool! What settings do you have for rubberband launch? Obviously they are tweaked from default so launch could be reliably detected.

sppnk, 2017-03-06:
@digitalentity, they are just defaults. These are my parameters for minitalon (I throw it by rubber):

The build is:
# INAV/CC3D 1.6.0 Feb  5 2017 / 11:16:23 (dcc163f)
mixer CUSTOMAIRPLANE

set nav_fw_launch_velocity = 300
set nav_fw_launch_accel = 1863
set nav_fw_launch_detect_time = 40
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set nav_fw_launch_motor_delay = 500
set nav_fw_launch_spinup_time = 100
set nav_fw_launch_timeout = 5000
set nav_fw_launch_climb_angle = 10

The launch works fine, I just have to tweak climb angle to my likes. I will raise launch_timeout to 7000 and launch_thr to 1950 (full thr).

raul-ortega, 2017-03-07:
As with RC1, I had the same problem with RC2 (the buzzer didn't help):
[https://www.youtube.com/watch?v=h6_dxqi4HC8](https://www.youtube.com/watch?v=h6_dxqi4HC8)

Not only me, other users of my community are reporting that the 99.99% of the launchings are unsuccessful, it doesn't matter which values they set in the nav_fw_launch_ parameters.

I'm very happy about reading that you have found the bug.

digitalentity, 2017-03-07:
@raul-ortega please let your fellow pilots know that 1.6.1 is going to be released in a day or so to fix this bug. In the meanwhile you are more than welcome to test the latest `development` branch.

closed 2017-03-27 by digitalentity