iNavFlight/inav #1321: FW ARMING feedback through servos movement

sppnk, 2017-02-23:
Many of us dont mount a beeper in our planes. With some models, the arming led can be seen through the foam, but there are other birds where this is not possible.

May be could be added a simple repetive sequence of servo movements when arming the plane.

Dronek, 2017-02-23:
In this context I'd like to suggest disabling (or a strong reduction of) servo throw in disarmed state, at least as an configurable option so adjusting and configuring servos could be still done in a disarmed state when the option is disabled.
Background: as of now, navigating an OSD menu by stick commands causes a lot of full throw servo activity.
It might also be an alternative to the repetitive sequence mentioned by @sppnk  because you could distinguish between armed and not armed easily by moving sticks and see if servos are moving or not.

sppnk, 2017-02-23:
@Dronek, you already can cut the outputs with inav configurator, that was an old issue solved.

oleost, 2017-02-23:
1. Just blip the throttle and you will figure out that your armed?

2. If using launch mode simple have passthrough selected, if launchmode is successful turned out it will start to stabilize ( move servos )

3. How much do you actually navigate through OSD menu, and it's not like it hurts the servos anyway if you move them while going through OSD?

Dronek, 2017-02-23:
> you already can cut the outputs with inav configurator

@sppnk: do you mean [this](https://github.com/iNavFlight/inav/pull/706)?
I thought, this would disable pwm output entirely regardless of arming state...
If it activates pwm output on arming, then it's the function, I'm looking for.

> How much do you actually navigate through OSD menu, and it's not like it hurts the servos anyway if you move them while going through OSD?

@oleost: much enough to ask for this feature :)
I use OSD menu to change parameters and when tuning PIDs I have the habit to activate the OSD menu in order to record my PID values and potential changes on DVR before the next flight. Certainly it's not a very important thing.

sppnk, 2017-02-23:
Yes, that function @Dronek.

digitalentity, 2017-02-24:
Reducing servo motion in disarmed state is probably a bad idea. If you accidentally disarm in air (happens sometimes, especially with stick arming) - you'll lose stabilisation completely until you re-arm .Current only I-term will be limited, but you'll still have at least some control over the machine and will be able to glide it down.

Dronek, 2017-02-24:
> Reducing servo motion in disarmed state is probably a bad idea. If you accidentally disarm in air ...

I agree, let's forget about reducing servo motion. This would probably cause a lot of confusion, too.
However, a sequence of servo movements on re-arming in mid-air is definitely something to avoid as well, now that I think about it.

oleost, 2017-02-25:
As previously stated I believe this is a bad idea.

There are ways of checking if your armed. As already explained.
Also buzzers are now tiny and costs next to nothing if you really want some feedback.

( You could even mount a display on your wing if you wanted to get it visually feedback. )

Dronek, 2017-03-02:
> Yes, that function @Dronek.

@sppnk : [this](https://github.com/iNavFlight/inav/pull/706) will actually disable all PWM output whatsoever regardless of arming state as I just tested on the bench. So it doesn't serve my purpose to disable servos on disarm only :(

digitalentity, 2017-03-30:
No solution has beed rached and personally I believe moving servos for any other purpose than steering is a bad idea (considering possibility to disarm/rearm in air).

closed 2017-03-30 by digitalentity

sppnk, 2017-03-30:
No problem. I agree, as I dont need this, because I am happy with my LTM telemetry showing all data now. :-)