iNavFlight/inav #1322: LTM stops when failsafe kicks in. 1.6 RC1

sppnk, 2017-02-23:
Hi, tested yesterday the following conditions:

- autolaunch set.
- plane armed ready to launch.
- LTM starts to work (as I have it configured to send when arming).
- I kick failsafe through switch or switching off my transmitter. This before launch the plane.
- LTM stops sending telemetry.

digitalentity, 2017-02-23:
If throttle was never increased failsafe will disarm instead of entering failsafe procedure. Try simulating a launch.

sppnk, 2017-02-23:
I dont see the led of the plane disarming ?¿

digitalentity, 2017-02-23:
Is LTM shared with MSP?

sppnk, 2017-02-23:
yes it is, because I use minimosd in the same port. Minimosd can read both protocolos at the same time, so I dont see difference in my lcd screen.

sppnk, 2017-02-23:
tested again; when I do a successful launch it remains sending LTM when I switch to failsafe. It stops only LTM if failsafe is activated during autolaunch, and also the launch is aborted (the motor stops). 

So closing this. Thanks.

closed 2017-02-23 by sppnk