iNavFlight/inav #1323: (Sensor failure detection) Compass is valid but data is not

giacomo892, 2017-02-23:
I'm doing some experiments to check edge cases of sensor failure.

I'm using a custom NAZE target (F1) running 1.6RC2 on a breadboard using an MPU6050 and HMC5883 as compass.

I want to simulate a compass failure. After some seconds after I unplug the compass from the board the MAG icon became RED and this is ok. 

As soon I replug the compass at it's place the icon became blue again (meaning MAG is healthy) but in sensors tab magnetometer values are stuck and those wrong, stuck, values are sent into the IMU giving totally wrong result messing the machine heading computation.

I know this is a very unfortunate fault. 
So i've tried to unplug only SDA or SCL (leaving mag powered and configured) and I can again encouter the same issue most of the times.

Can we do something more to check the actual health of the mag and re-init it?  like heading is changing but mag value are not.

I'm here if needed

digitalentity, 2017-02-23:
We have an issue for this check already - https://github.com/iNavFlight/inav/issues/991

giacomo892, 2017-02-23:
If you want, you can close this issue. 
Btw. am I wrong saying that navigation should work even after a compass failure (at least for emergengy RTH) since heading computed with gyro is quite stable and the quad is facing home during RTH so also GPS heading could be used.

digitalentity, 2017-02-24:
@giacomo892 I'd rather keep it open for the time being.

After you acquire true heading you no longer need compass - gyro drift is usually quite small (few degrees a minute). On quads GPS heading can't be used because quad can fly in any direction and GPS heading indicate direction of movement and not the direction where machine is facing.

giacomo892, 2017-02-24:
Ok @digitalentity .
I was meaning something a little different. 
Put we have just detected a compass failure on MC. Excluding it from heading computation we should be able to rotate the MC facing home direction. At this point why cannot be use the GPS heading to navigate home? (assuming the quad is capable of maintaining the heading)

digitalentity, 2017-02-24:
@giacomo892 I understand what you are suggesting - that will be a totally different mode of operation for quads. Usually quad direction of flight is disconnected from heading. It can start flying home tail tail first and turn in correct direction in process. That's not possible when using GPS-only heading - we'll have to modify the whole procedure for multirotor flight control to be able to fly like this.

giacomo892, 2017-02-24:
Ok glad you have locked my point :) Thanks for your effort being put in INAV.

digitalentity, 2017-02-24:
@giacomo892 I have something that will allow compass-less navigation on the multirotors - this feature is currently in development and testing. Will be a killer feature.

giacomo892, 2017-02-24:
Oh! What a news !!!! I guess I will have to upgrade to F3 :D :D :D :D Keep going!

liguo-git, 2017-03-03:
My compass is sometimes not recognition，If the recognitionis always  is East。

andre92, 2017-03-29:
@digitalentity if a multicopter flies at sufficient speed, can the heading be determined from gps course and attitude?

digitalentity, 2017-03-29:
@andre92 we can make assume heading if attitude is not changing too much. I have some interesting code it the works to have a backup system for compass - need to perfect it first.

closed 2023-10-27 by DzikuVx