iNavFlight/inav #1326: Missing data whith integrated OSD in iNav 1.6-RC2

raul-ortega, 2017-02-24:
By default we have this:

set osd_gps_lon = 2263
set osd_gps_lat = 384
set osd_home_dir = 366
set osd_home_dist = 2081

Those fields doesn't appear on screen or they are not in the correct position. Setting them with differents values solves the problem.

In the configurator they also doesn't appear, perhaps as consecuence of the name of the  cli parameters. The sufix _pos are missing on them.

raul-ortega, 2017-02-28:
Solve in final version 1.6, but Home Direction togle button activates Home Distance data, and viceversa. [See video](https://youtu.be/em8tzH-JVjc)

DzikuVx, 2017-03-01:
OK, I'm closing this one. Mixed OSD elements will be fixed in separate ticket

closed 2017-03-01 by DzikuVx