iNavFlight/inav #1330: angle mode  while throttle on - error

kozak71, 2017-02-25:
hi, i have issue with servo and throttle in angle mode.

i'm using flying wing mode.
while i armed motor, and put angle mode, when push motor stick ( above 1100) left or right aileron moves slow to max up/down. 
when throttle stick moves down, to stop. ailerons go back to neutral. 

what is wrong ? it is only in angle mode.

oleost, 2017-02-25:
XSounds normal. This is I gain starting to work when you throttle up. Your fixed wing is probably not 100% flat and the flight controller tries to level the plane, and since nothing is happening it will give more and more throw on ailerons.

closed 2017-02-25 by oleost

kozak71, 2017-02-25:
No, wing is leveled, it is on the floor. 

1. armed motor
2. put angle on
3. move throttle stick up
4. ailerons goes down
3. put throtle stick to 0
4. ailerons go to neutral

see video : https://www.youtube.com/watch?v=ZlU_MkD2RdE&feature=em-upload_owner#action=share

When i change PID on I to 0, it is ok, but this is not a solution... 
On the other soft ( not inav) everything is ok.

digitalentity, 2017-02-25:
@kozak71 you can't place the wing level without any error. Even 0.0001 deg error will be treated as *non-level* by the software. Please go and fly your wing.