iNavFlight/inav #1331: angle level error

kozak71, 2017-02-25:
i'm using flying wing mode.
while i armed motor, and put angle mode, when push motor stick ( above 1100) left or right aileron moves slow to max up/down.
when throttle stick moves down, to stop. ailerons go back to neutral.

what is wrong ? it is only in angle mode.

 wing is leveled, it is on the floor.

armed motor
put angle on, ailerons are on neural
move throttle stick up
ailerons goes down
put throtle stick to 0
ailerons go to neutral
In Acro mode all is ok, it is only in ANGLE
see video : https://www.youtube.com/watch?v=ZlU_MkD2RdE&feature=em-upload_owner#action=share

throtle max, :
http://www.fotosik.pl/zdjecie/usun/CfsaOUX~g798pKN~0hKkWueJzaM7Cm

gyro is ok, see below photo :
http://www.fotosik.pl/zdjecie/usun/geMt2ubLQ1CBWJD8hRpexkDWu98h5V

When i change PID on I to 0, it is ok, but this is not a solution...
On the other soft ( not inav) everything is ok.

andre92, 2017-02-25:
Try enabling airmode, that will prevent the PID controller from resetting to center.

kozak71, 2017-02-25:
I did this but without positive result :(

kozak71, 2017-02-25:
I tried angle + airmode , correct ?

andre92, 2017-02-26:
Could you post a cli dump, including board and version? At this point I am not sure what is going wrong.

kozak71, 2017-02-26:
# version
# INAV/KFC32 1.6.0 Feb 17 2017 / 18:13:45 (1c373bf)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -RX_ELERES
feature VBAT
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD
feature RX_ELERES

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 430 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-02-27:
This is normal behavior. You can't place the craft 100% level - even 0.01deg error is non-level and FC is trying to correct for it. Try flying it.

closed 2017-02-27 by digitalentity